microsoft/Azure-Kinect-Sensor-SDK #1361: @HaipengU when i config : deviceConfig.color_format = K4A_IMAGE_FORMAT_DEPTH16; but  NativeKinectDevice.start_cameras(&deviceConfig); it is not work otherwise deviceConfig.color_format = K4A_IMAGE_FORMAT_COLOR_BGRA32 .,I want to resolve it ,who can help me 3Q

UEBoy2019, 2020-09-24:
@HaipengU Thank you for opening the git hub issue, yes, the depth image contains width * height of uint16_t pixel values, e.g. you can see the documentation of the image format description [here](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___enumerations_gabd9688eb20d5cb878fd22d36de882ddb.html#gabd9688eb20d5cb878fd22d36de882ddb) (K4A_IMAGE_FORMAT_DEPTH16). The buffer store the image as row-major order, you can read the depth from the buffer by first casting the buffer to uint16_t*, then for a pixel with index `[i, j]`, you can use `j * width + i` to get the depth value from the uint16_t* depth buffer

_Originally posted by @rabbitdaxi in https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/968#issuecomment-566213950_

UEBoy2019, 2020-09-24:
### source 
k4a::device NativeKinectDevice = k4a::device::open(DeviceId);
k4a_device_configuration_t deviceConfig;
deviceConfig = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
deviceConfig.depth_mode = K4A_DEPTH_MODE_NFOV_UNBINNED;
deviceConfig.color_format = **K4A_IMAGE_FORMAT_DEPTH16;**
NativeKinectDevice.start_cameras(&deviceConfig);

but do this ,it will work
k4a::device NativeKinectDevice = k4a::device::open(DeviceId);
k4a_device_configuration_t deviceConfig;
deviceConfig = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
deviceConfig.depth_mode = K4A_DEPTH_MODE_NFOV_UNBINNED;
deviceConfig.color_format =**K4A_IMAGE_FORMAT_COLOR_BGRA32;**
NativeKinectDevice.start_cameras(&deviceConfig);

qm13, 2020-09-25:
@UEBoy2019 k4a_image_format_t values are primarily used to indicate the format of the returned image. The COLOR values also are used as values for setting the color camera capture format. K4A_IMAGE_FORMAT_DEPTH16 is only used in indicating the format of a returned depth image. You second code example is correct. You can also use K4A_IMAGE_FORMAT_COLOR_MJPG and K4A_IMAGE_FORMAT_COLOR_NV12.

UEBoy2019, 2020-09-29:
have a good happy to u! 3Q

UEBoy2019, 2020-09-29:
> @UEBoy2019 k4a_image_format_t values are primarily used to indicate the format of the returned image. The COLOR values also are used as values for setting the color camera capture format. [...]

have a good happy to u! 3Q

closed 2020-09-29 by UEBoy2019