microsoft/Azure-Kinect-Sensor-SDK #1364: I want to use tansformation depth image to color camera ,but failed

UEBoy2019, 2020-09-30:
```
.h  
k4a_calibration_t             cali_t;
k4a::transformation        trans;
k4a::calibration               sensorCalibration;

.cpp
NativeKinectDevice               = k4a::device::open(DeviceId);
deviceConfig                         = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
deviceConfig.depth_mode    = K4A_DEPTH_MODE_NFOV_UNBINNED;
deviceConfig.color_format    = K4A_IMAGE_FORMAT_COLOR_BGRA32;
deviceConfig.color_resolution           = K4A_COLOR_RESOLUTION_1080P;
deviceConfig.synchronized_images_only = true;
deviceConfig.wired_sync_mode = k4a_wired_sync_mode_t::K4A_WIRED_SYNC_MODE_STANDALONE;
NativeKinectDevice.start_cameras(&deviceConfig);

sensorCalibration  = NativeKinectDevice.get_calibration(deviceConfig.depth_mode, deviceConfig.color_resolution);
NativeBodyTracker   = k4abt::tracker::create(sensorCalibration);
k4a_device_get_calibration(NativeKinectDevice.handle(), deviceConfig.depth_mode, deviceConfig.color_resolution, &cali_t))
trans = k4a_transformation_create(&cali_t);

k4a::image tansformedImage;
k4a::image img = sensorCapture.get_color_image();	
k4a::image depth = sensorCapture.get_depth_image();
trans.depth_image_to_color_camera(depth, &tansformedImage);
//func,  it will go to :  catch (k4a::error captureError)

```

wes-b, 2020-10-02:
tansformedImage is not initialized property, try: 
> tansformedImage = trans.depth_image_to_color_camera(depth);

UEBoy2019, 2020-10-09:
> tansformedImage is not initialized property, try:
> 
> > tansformedImage = trans.depth_image_to_color_camera(depth);

sorry to tell u,  it will go to :  catch (k4a::error captureError),   still.

UEBoy2019, 2020-10-09:
> tansformedImage is not initialized property, try:
> 
> > tansformedImage = trans.depth_image_to_color_camera(depth);

3Q, u are right ,use this func, 
the answer question  maybe is what i use c type to define cali,otherwise cpp type;
last do this , it will work well ,
k4a::calibration cali=k4a::transformation(cali)；
NativeKinectDevice.start_cameras(&deviceConfig)；
k4a::image depth = sensorCapture.get_depth_image();
k4a::image tansformedImage==trans.depth_image_to_color_camera(depth)；

closed 2020-10-14 by UEBoy2019

JSCG95, 2022-10-26:
Hello, I'm trying to adapt depth_image_to_color_camera transformation but using a python SDK wrapper. My question is why depth_image_to_color_camera(depth) is just using the depth image as an argument without using the transformation handle as the first argument?

VegetableWithChicken, 2022-10-26:
您好，我一般不常用邮箱，有重要邮件，请QQ我下，谢谢!