microsoft/Azure-Kinect-Sensor-SDK #1365: several body tracking APIs missing const qualifier, prevents const objects/variables

diablodale, 2020-10-01:
Multiple C++ APIs for body tracking are missing the `const` qualifier. By not having `const` on these c++ member functions, it prevents the use of const objects. Such objects are desired by style, coders, and compliers.

I have so far found the following. There may be others. These all return POD. I don't readily see any reason to prevent them being `const`.

```c++
k4abt::frame
  get_num_bodies()
  get_body_skeleton()     both signatures
  get_body_id()
  get_body()
```

## Setup

Microsoft Windows [Version 10.0.19042.546]
Azure Kinect SDK 1.4.1
Body Tracking SDK 1.0.1
Visual Studio Community 16.7.2
RGB camera firmware:      1.6.110
Depth camera firmware:    1.6.79

## Repro

An example of the issue can be reproduced in two lines...

```c++
const k4abt::frame btframe(nullptr);
const auto numBod = btframe.get_num_bodies();
```

### Result

Compiler errors and failed compile.

```
'uint32_t k4abt::frame::get_num_bodies(void) noexcept': cannot convert 'this' pointer from 'const k4abt::frame' to 'k4abt::frame &'
'numBod': an object of const-qualified type must be initialized
```

## Expected

Clean compile. No errors.

## Notes

Lets look at one of the APIs `k4abt::frame::get_num_bodies()`

* not qualified `const`
* returns POD of uint32_t
* no need to add another reference on to the frame handle since the `k4abt::frame` of that member function
  already holds a reference
* the code in that member function is a single line `return k4abt_frame_get_num_bodies(m_handle)`
* that C API's param is non-const

Somewhat related: the C APIs that implement the functionality in these member functions can likely have const params themselves. They are not going to be changing the frame handle itself.

diablodale, 2020-11-28:
Additionally

```c++
k4abt::tracker
  pop_result()             both signatures
  set_temporal_smoothing()
  shutdown()
```

I do hesitate slightly in one area. That last one `shutdown()` can technically be const.  The `tracker` c++ object doesn't change since it contains `m_handle` and that is used to call the C api k4abt_tracker_shutdown(m_handle) and no part of the C++ object changes. The thing to which the handle points does change. But not the handle itself, and not `tracker`. So technically, it can be const. However since shutdown() causes a drastic *logical* change to the logical tracker (not the technical `k4abt::tracker`), I can imagine a viewpoint that making it non-const conveys such a logical change.

As a rigorous coder, I suggest make them consistently `const` in all cases possible. Simultaneously, I acknowledge there is another possible viewpoint.

diablodale, 2021-04-17:
This issue is partially fixed in Body Tracking SDK v1.1.0.
The apis still capable of `const` are
```
enqueue_capture()
pop_result()             both signatures
set_temporal_smoothing()
shutdown()
```

qm13, 2022-03-21:
Fixed

closed 2022-03-21 by qm13