microsoft/Azure-Kinect-Sensor-SDK #1367: Body tracking not working from Unity on Windows 2004

DanielEliasib, 2020-10-01:
**Describe the bug**

I'v being fighting with this bug for several day. I have a Unity project that uses the C# wrapper. One day it randomly started to crash all the time and after creating a fresh one and debugging I was able to find that the problem came from: `Microsoft.Azure.Kinect.BodyTracking.Tracker.Create();`. In the fresh Unity project it would log **K4A_RESULT_FAILED**.

Then I realised the problem started after windows made me update to 2004, I reverted the update and now the same code works as expected. 

It's a tricky bug because running the body tracking example that comes with the sdk works fine, so this may be something related to Unity as well, but only this was failing. Also is worth noting that using a version of my code without the body tracking worked fine.

**To Reproduce**
I can't reproduce it anymore because I reverted my windows installation to 1903 and never created a step by step process because I was just trying to fix it. Maybe on the weekend I'll try with my home PC.

But If I were to give a list of steps It'd be:

1.- Create a Unity 2019.3 or 2019.4 project,
2.- Create a script that calls `Tracker tracker = Tracker.Create(device.GetCalibration(), new TrackerConfiguration() { ProcessingMode = TrackerProcessingMode.Gpu, SensorOrientation = SensorOrientation.Default });`, using the body tracking sdk.

You'll need to add the relevant dll for this.

I'll update this section if I get to reproduce it this weekend. 

**Expected behavior**
Tracker.Create(); is expected to initialized body tracking. 

**Logs**
Don't have logs, and can't reproduce anymore because I changed Windows versions. 

**Desktop (please complete the following information):**
 - OS with Version: Windows 2004
 - SDK Version:  1.0.1
 - Firmware version:  1.6.75

**Additional context**
Maybe it's one of those isolated case of weird bugs that happen when windows updates.

benMOOVENCY, 2020-10-12:
Hello @DanielEliasib  you could look these issues 
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/932.

there is maybe a pb with the onxruntime.dll

henryo12, 2020-10-23:
Are there any update on this? I am experiencing the same issue. My project runs as expected inside the Unity Editor, but when I build the code, the exe fails on the Tracker.Create(). This also happens when I start fresh with the sample_unity_bodytracking example. Do I need to add any dll files to the build folder (other than those placed in build/Data/Plugins), or is this likely related to the issue with onnxruntime.dll?

To recreate:
1. Setup [sample_unity_bodytracking](https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples/sample_unity_bodytracking)
2. Build the Kinect4AzureSampleScene (Windows x86_64)
3. Run the executable (UnityCSharp.exe)

Details:
- Unity 2019.2.17 and 2019.4.5. 
- Windows 10 Edu 1909.
- SDK Version: 1.0.1

DanielEliasib, 2020-10-25:
Hi, I haven't been able to test this solution, this weekend I unpaused the updates to try this out but it seems that now Windows Update can't find the 2004 update that was causing problems, or any newer updates. I guess I have to wait until the October update is pushed to my device.

DanielEliasib, 2020-10-29:
I can confirm that the solution listed above seems to work, I'm on build 20241.1005 and I can run the body tracking with no issue.

closed 2020-11-30 by qm13