microsoft/Azure-Kinect-Sensor-SDK #1372: (HIP_LEFT,HIP_RIGHT)'s orientations are actually (KNEE_LEFT,KNEE_RIGHT)'s

knat, 2020-10-04:
```C++
k4abt_skeleton_t skeleton = ...;
k4a_quaternion_t const& hipLeftQ = skeleton.joints[K4ABT_JOINT_HIP_LEFT].orientation;
k4a_quaternion_t const& hipRightQ = skeleton.joints[K4ABT_JOINT_HIP_RIGHT].orientation;
```
They are orientations of bones marked with red:
[image]
[image]

It's VERY confusing.

qm13, 2020-10-05:
We are confused as to what you are asking? Are you asking why the hip orientation is the same as the knee orientation?

knat, 2020-10-05:
Hi,
```C++
k4a_quaternion_t const& SPINE_NAVAL_Q = skeleton.joints[K4ABT_JOINT_SPINE_NAVAL].orientation;
```
`SPINE_NAVAL_Q` is the orientation of the bone that points to SPINE_NAVAL joint:
[image]

```C++
k4a_quaternion_t const& HIP_LEFT_Q = skeleton.joints[K4ABT_JOINT_HIP_LEFT].orientation;
```
Is `HIP_LEFT_Q` the orientation of the bone that points to HIP_LEFT joint? No, it's the orientation of the bone that points to KNEE_LEFT joint.

It's not a bug, but very confusing. It takes me hours to figure it out.

closed 2020-11-05 by knat