microsoft/Azure-Kinect-Sensor-SDK #1373: add check for nullptr handle in body track c++ `shutdown()`

diablodale, 2020-10-04:
The `k4abt::shutdown()` api doesn't check its `m_handle` before calling the C api `k4abt_tracker_shutdown()`. This is a minor inconsistency as compared to its sibling `k4abt::shutdown()` which *does* check its handle with `if (m_handle != nullptr)`.

## Ask

I request the handle test be added to `k4abt::shutdown()` so to be in alignment with its sibling `shutdown()` and remove boilerplate from user code.

## Fix

Add the handle test to the C++ body track header like the following. While it is not necessary, I also added `const` to the function signature since it does not modify the class variables.

```c++
void shutdown() const noexcept
{
    if (m_handle != nullptr)
    {
        k4abt_tracker_shutdown(m_handle);
    }    
}
```

## Alternatives

In a scenario of serial shutdown and destroy...

```c++
bodyTracker.shutdown();
bodyTracker.destroy();
```

User code can use a one line if test

```c++
if (bodyTracker) bodyTracker.shutdown();
bodyTracker.destroy();
```

HlibKazakov2000, 2021-04-13:
> Отправлено: 13 апреля 2021 г. 7:52
> Кому: microsoft/Azure-Kinect-Sensor-SDK ***@***.***>
> Копия: Hlib Kazakov (Hi-Tech Talents LLC) ***@***.***>; Assign ***@***.***>
>
> Assigned #1373<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fmicrosoft%2FAzure-Kinect-Sensor-SDK%2Fissues%2F1373&data=04%7C01%7Cv-hkazakov%40microsoft.com%7C113c01235d8441753aa708d8fe404646%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C637538899322824725%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=WKoib6AbRlHE1jGWTMhvasNPhbXnSbyCU83fta%2FBKAw%3D&reserved=0> to @HlibKazakov2000<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FHlibKazakov2000&data=04%7C01%7Cv-hkazakov%40microsoft.com%7C113c01235d8441753aa708d8fe404646%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C637538899322834685%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=eIIw3In%2Fmo33isS6uYt4DeKwrlEKqek9ByVsGlsvvtE%3D&reserved=0>.

Should I wait for a new release before resolving the issue or is it fine to reply that it will be fixed in an upcoming release and resolve?

diablodale, 2021-04-14:
Hi. I prefer to leave it open. I verify fixes for all the bugs that I open and aren't closed (against my wishes). I have so many open issues, that I must have tools (like open github issues) to manage them.

This allows somewhat the better phase changes of:  open -> resolved -> closed.
But github is lacking that middle phase. So open is the only realistic alternative.

qm13, 2022-03-21:
Fixed

closed 2022-03-21 by qm13