microsoft/Azure-Kinect-Sensor-SDK #1374: OpenCV KinectFusion Sample - Questions

jessekirbs, 2020-10-04:
Hello,

I have a couple questions about this software that I can't find the answers to.

- Can this be used with multiple Kinects for Azure?
- Does this software handle the calibration of the Kinects? I don't quite understand what it means by 'how to feed calibration'. Do the Kinects need to be calibrated with other software first?
- Can this software be used to merge and mesh pointclouds that have been previously recorded? Or is it only realtime? I'd like to record a sequence of multiple aligned pointclouds and then use this software to merge and mesh them. Is that possible?

Thank you!

qm13, 2020-10-05:
@jessekirbs the KinectFusion example only uses one Azure Kinect DK. Please look at the [green_screen](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/green_screen )sample for multi-camera support including calibration. You should be able to use learnings from the green_screen sample to enhance the KinectFusion example for multiple AKDKs.

jessekirbs, 2020-10-06:
@qm13 Appreciate the reply! Are you able to expand on this:

Can this software be used to merge and mesh pointclouds that have been previously recorded? Or is it only realtime? I'd like to record a sequence of multiple aligned pointclouds and then use this software to merge and mesh them. Is that possible?

Thank you!

qm13, 2020-11-30:
The sample is only realtime. That said the process of merging the two point clouds is the same whether realtime or recorded.

closed 2021-01-08 by qm13