microsoft/Azure-Kinect-Sensor-SDK #1375: The device keeps blinking in device manager

greatwallet, 2020-10-05:
## System
* OS: Windows 10 
* CPU: Intel i7 8700
* RAM: 16G

## Problem:
When I tried to connect Kinect DK to PC for second time, the `Azure Kinect Viewer`  does not detect any device. 
And the LED light white with a frequent pause (like every 2-3 seconds), while the device manager detects `4K Camera` but kept freshing up with same frequency.

Have tried several PCs, and same problem occurs.

<b>Note: </b> For the first time, the camera ran smoothly, but suddenly the  `Azure Kinect Viewer` reported a bug , somthing like "no device is detected at usb port" and the software got dead.

## Tried solutions (in vain)
I used a phone charger cable (type-c to usb) to connect PC with Kinect DK, but in vain. (`4K Camera` does not even appear on device manager)

## Expected advice
* What exact kind of cable should I use, if the problem really arises from the cable? (A product page link would also be appreciated!)
* Is it likely that the camera itself could be damaged inside?
* Any other suggestions?

qm13, 2020-10-05:
@greatwallet the blinking white LED indicates that the device is connected to a USB2 interface. It requires USB3. Can you please verify that you are connecting it to a USB3 interface? Please use the supplied cables. If can confirm you are using a USB3 interface and it still does not work then please test with an alternative USBA - USBC power+data cable or a USBC-USBC power+data cable.

Phone charging cables do not support data and will not work with Azure Kinect.

closed 2020-11-30 by qm13

nguyenv7, 2021-03-16:
Sorry to reopen this issue.
I am getting the same problem now. The device blinking and Device Manager recognized the Azure Kinect but not constantly. It is "refreshed" the detection of this device in the same frequency as the white blinking.
This behavior happens regardless of my host devices: MS Surface book 2, Lenovo T470, my Linux PC. All of them are equipped with USB 3.0 port, try all possible USB 3.0 - USB-C cable, including the provided cable from the Azure Kinect package.

I can not reset the Kinect as the instruction as well.

greatwallet, 2021-03-17:
> Sorry to reopen this issue.
> I am getting the same problem now. The device blinking and Device Manager recognized the Azure Kinect but not constantly. It is "refreshed" the detection of this device [...]

Might be breakdown in device itself. 
In my case, I have contacted the sales' and get a new device.

nguyenv7, 2021-03-17:
Thanks @greatwallet :), I will contact the seller to get the new one.