microsoft/Azure-Kinect-Sensor-SDK #1376: Writing image buffer data to JPEG file

unbeschwert, 2020-10-08:
**Describe the bug**
I have written the following c++ code to capture the color image and write the image buffer data to the JPEG file. 
`
        k4a::image rgb_image_handle = capture_handle.get_color_image();
	if (rgb_image_handle.is_valid()) {
		// code for storing the buffer in an image file
		uint8_t* image_buffer = rgb_image_handle.get_buffer();
		size_t image_buffer_size = rgb_image_handle.get_size();
		const std::string file_name = "hello_rgb.jpeg";
		std::ofstream file_object(file_name, std::ios::out);
		file_object.write(reinterpret_cast<char*>(image_buffer), 
			image_buffer_size);
		file_object.close();
	}
	else {
		std::cout << "Invalid image!\n";
		exit(1);
	}
`

I start the device cameras using the following configuration:
`
        config.color_format = K4A_IMAGE_FORMAT_COLOR_MJPG;
	config.color_resolution = K4A_COLOR_RESOLUTION_720P;
	config.depth_mode = K4A_DEPTH_MODE_NFOV_UNBINNED;
	config.camera_fps = K4A_FRAMES_PER_SECOND_30;
	config.synchronized_images_only = true;
	config.depth_delay_off_color_usec = 0;
	config.wired_sync_mode = K4A_WIRED_SYNC_MODE_STANDALONE;
	config.subordinate_delay_off_master_usec = 0;
	config.disable_streaming_indicator = false;
`

However, when I open the JPEG file, I dont see anything. I get the following error "File can't be opened". I did try to write the buffer data to console and it appears to be correct. 

**To Reproduce**
1. Install latest release of the sensor SDK.
2. Execute my program 
3. Get an output JPEG file that doesn't open. 

**Expected behavior**
An image that captures the area where I am pointing my camera. 

**Desktop (please complete the following information):**
 - Windows 1909
 - SDK Version:  1.4.1
 - RGB camera firmware:      1.6.102
 - Depth camera firmware:    1.6.75
 - Audio firmware:           1.6.14

Any help is highly appreciated.

UEBoy2019, 2020-10-09:
i think u should 
first  file_object.open(),
and just call once to save image,
then file_object.close().

unbeschwert, 2020-10-09:
Hi @UEBoy2019 I dont think that's the problem. I digged through the microsoft STL and found that the [write](https://github.com/microsoft/STL/blob/a6c10429b6326aa92b3e410ed6a1f3fa13ffa571/stl/inc/ostream#L526) function uses [sputn](https://en.cppreference.com/w/cpp/io/basic_streambuf/sputn) function which writes to the stream buffer which is managed by the file_object. Hence, I think we dont need open and close functions.(my code is an iteration of something I have been rewritting. Hence it contains the close method.)

For e.g. this code
```
#include <iostream>
#include <fstream>
#include <cstring>

int main(int argc, char* argv[]) {
    std::string file_name = "hello";
    const char* to_write = "Hey Man how are you ?";
    std::ofstream file_object{file_name, std::ios::out};
    file_object.write(to_write, strlen(to_write));
    return(0);
}
```
works. 

This is what is confusing me. I dont know if the problem lies with my code or with the output of the camera unfortunately.

qm13, 2020-10-13:
The fix is like to add a binary format to ```ofstream```. 

                  std::ofstream file_object("hello_rgb.jpeg", std::ios::out | std::ios::binary);
                  file_object.write(reinterpret_cast<char*>(image_buffer), 
                            image_buffer_size);
                  file_object.close();

unbeschwert, 2020-10-14:
@qm13 Thank you. That worked. I will close this issue.

unbeschwert, 2020-10-14:
Hi @qm13  I am reopening this issue because I have found some issues with the image that is being produced.
# Issue 1: Distorted image
Sometimes the image generated using the above code looks distorted. Belowimage is a pretty good example
[image]
## Note: This happens only few times. Not regularly. Could you please provide some pointers as to why this is happening ?

# Issue 2: OpenCV unable to read the image
The following code doesn't work as expected. 
``` cv::Mat image = cv::imread(image_path, cv::IMREAD_COLOR); ```
Unfortunately, OpenCV on windows doesn't throw any error. I did copy the file to a debian server and there I got the following error:
```Corrupt JPEG data: premature end of data segment``` 
which leads to me believe that the jpeg data sent by the camera is corrupted. Could you please look into this ?

qm13, 2020-10-15:
This may be the cause of your issue. Make sure you [use flush](https://stackoverflow.com/questions/5036878/why-does-ofstream-require-a-flush) before closing the file.

unbeschwert, 2020-10-15:
@qm13  Yes this worked. Thank you! Is there a place where these "gotchas" are documented ? These minor issues are increasing dev time more than I would like to admit.

closed 2020-10-15 by unbeschwert