microsoft/Azure-Kinect-Sensor-SDK #1378: Manual exposure has much lower brightness than auto exposure

ChristopherRemde, 2020-10-09:
**Describe the bug**
When a kinect is set to manual exposure mode, and the device is then closed and opened again, the brightness of the RGB image is much lower after the restart. Even if the exposure step is turned to it's maximum value, the image doesn't get as bright as when auto exposure is enabled. Maybe related to #1352 and #549 

**To Reproduce**
1. Open Kinect Viewer
2. Open a Kinect Device and start it
3. Set the exposure mode to auto. 
4. Set the exposure mode to manual and turn the exposure time to its highest possible value. 
5. Note the RGB image brightness. You should be able to see a bright and nicely lit image.
6. Stop the device
7. Open the device again. Set the exposure time to its highest possible value. Now the RGB image is much darker than before. Even on the highest exposure time, the image is not as bright as when using auto exposure
8. When exposure is set to auto and then manual again, the same exposure times as before produce a much brighter image

**Expected behavior**
Image from 4 & 7 should have the same brightness

**Screenshots**
Image brightness when the camera is started in auto exposure mode, then set to manual exposure and a exposure time of 33330us:

[image]

Image brightness when the camera is started in manual exposure mode, with a exposure time of 33330us:
 
[image]

**Desktop (please complete the following information):**
 - OS with Version: Windows Version 10.0.19041
 - SDK Version:  v1.4.1
 - Firmware version: RGB Camera: 1.6.110, Depth Camera: 1.6.79, Audio Firmware: 1.6.14

aashutosh1997, 2021-04-27:
@jaygullapalli @qm13 any update on this? Need to get this fixed asap, willing to contribute if it could expedite things. Thanks!

dinvincible98, 2021-07-09:
@jaygullapalli I also encounter this auto-exposure problem. Any updates to fix it?