microsoft/Azure-Kinect-Sensor-SDK #1380: it looks like distorted Depth Image of Object

yeongwoonIm, 2020-10-12:
Depth Image of Object looks like distorted.
[image]
[image]

The other TOF Camera doesn't look like distorted.. Intel TOF, CubyEye TOF,  etcs..

Could you fix that Depth Image of Object undistorted ?

When I call this function. it already has been undistorted it. and I use it. 

>  Capture capture = device.GetCapture();

I searched issues. but there is no solutions that I want it.

Thank you.

UEBoy2019, 2020-10-12:
Capture cap=k4a::capture::create();
device.get_capture(cap);
k4a::image color=cap.get_color_image();
k4a::image depth=cap.get_depth_image()
k4a::image ir=cap.get_ir_image()

qm13, 2020-10-12:
Could you please share the original image prior to un-distortion.

yeongwoonIm, 2020-10-13:
> Could you please share the original image prior to un-distortion.

Those Images are Original Image. it just coded.
There is viewer Image
[image]

yeongwoonIm, 2020-10-13:
> Capture cap=k4a::capture::create();
> device.get_capture(cap);
> k4a::image color=cap.get_color_image();
> k4a::image depth=cap.get_depth_image()
> k4a::image ir=cap.get_ir_image()

it;'s C++ Code.
I code C#. and it may be same.
Capture capture = device.GetCapture();
short[] realtime_depth = capture.Depth.GetPixels<short>().ToArray();
short[] realtime_activeIR = capture.IR.GetPixels<short>().ToArray();

rapidengineering, 2020-10-13:
#1238  had the same issue. It was believed that it's an hardware error, but it's not.

yeongwoonIm, 2020-10-13:
> #1238 had the same issue. It was believed that it's an hardware error, but it's not.

#1238 is closed...
is it solve?? or still processing??

rapidengineering, 2020-10-13:
The depthimage is still distorted, but in our usecase I found a workaround.

In our case we want to detect objects in the color images and use that detection to filter the pointcloud.

_First we crop out the depthimage out of the color image._
so for example if you use NFOV Unbinned with 720p color resolution.
-Color =1280x720
-Depth=640x576

You crop your color image to:
X=1280/2-640/2
Y=720/2-576/2
Width=640
Heigth=576

_Second step is getting the boundingboxes_

_Third step is use that boundingbox to filter the pointcloud._
To do this you have to apply the 6 degree rotation to your boundingboxes  (see documentation).

I dont know if this is helpfull because the distortion is not really gone.
[image]

qm13, 2020-10-13:
The wide field of view of the AKDK depth sensor meant it was more optimal to use an f-theta (fisheye) rather than a rectilinear lens (e.g. a typical photographic camera or most machine vision cameras). This results in barrel distortion that results in some flat objects appearing curved, similar to a color image taken through a fish-eye lens. Most competitive products do not have such a wide field of view and do not need to make this trade-off. 

Many image processing and machine learning algorithms make the assumption of a rectilinear lens or an explicitly undistorted image. The RGB sensor on the Kinect for Azure has rectilinear distortion, however the raw depth data provided through the API is not automatically resampled to remove the barrel distortion. The Sensor SDK supports mapping/resampling the depth data from the raw images to the RGB images if that is useful (using ```k4a_transformation_create``` and ```k4a_transformation_depth_image_to_color_camera functions```), which will remove the distortion but may crop/not provide the exact resolution that you need.

An alternative approach is to use the calibrated camera intrinsic (get using ```k4a_device_get_calibration```) with OpenCV or alternative CV library to undistort the image.

yeongwoonIm, 2020-10-14:
> The depthimage is still distorted, but in our usecase I found a workaround.
>
> In our case we want to detect objects in the color images and use that detection to filter the pointcloud.
>
> _First we [...]

Wow~ 
You have done great job !!
it's really helpfull

yeongwoonIm, 2020-10-14:
> The wide field of view of the AKDK depth sensor meant it was more optimal to use an f-theta (fisheye) rather than a rectilinear lens (e.g. a typical photographic camera or most machine vision [...]

Thank you for good advice.
the distortion almost removed.
[image]

and I want to ask one more things.

How can I get IR data?
I can't see code of getting IR data in C# API.

closed 2020-10-21 by yeongwoonIm