microsoft/Azure-Kinect-Sensor-SDK #1382: convert k4a::image to cv::Mat ,and it will stutter after  about a minute later

UEBoy2019, 2020-10-14:
**Environment:**
win10;
VS2017,
i7-9700KF,
RAM:48G,
2080TI；
UnrealEngine4.25.3

**question One:**
i convert k4a::image to Mat,and Mat to Texture2D of unreal engine 4.25.3,but it will  cyclical stutter after about a minute later,
i think the problem is what AzureKinectDK will Memory recycling when RAM is almost full or cycle time 
right ?please help me !!!

**question Two;**
do this:
k4a::image tansformedImage,transformedRGB;
k4a::image img = sensorCapture.get_color_image();
k4a::image depth = sensorCapture.get_depth_image();
tansformedImage=trans.depth_image_to_color_camera(depth);
cv::Mat ColorFra= K4a2RGBMat(img);
cv::Mat TranFra = K4a2DepthMat(tansformedImage); 

/*UnrealEngine render thread func*/
AsyncTask(ENamedThreads::GameThread, [&]()
{
/*Test the following two functions*/
Mat2RGBTexture(ColorFra);  //**only run this func,** the system-Task Manager - Performance - memory will increase from 7G to 27G,and  cycle;
Mat2TransformedTexture(TranFra);//**Only run this func**, the sustem-Task Manager - Performance - memory will increase from 7G to 17G,and cycle;
**_### important:_**
run two functions ,the RAM will full ,and almost Stuck,the RAM will increase 7G to 50G,and more; 
});

two question ,thank u!!!

qm13, 2020-10-15:
The machine looks well capable of running this application. To check please run k4aviewer with color and depth on for at least an hour. If all ok then you need to start peeling the onion. Comment out all Unreal and CV functions. Does it run ok? Add back CV. Does it run ok? etc...
Without seeing all the code for the get capture loop it is hard to tell where the issue is.

UEBoy2019, 2020-10-16:
> The machine looks well capable of running this application. To check please run k4aviewer with color and depth on for at least an hour. If all ok then you need to start peeling the onion. Comment [...]

closed 2020-10-16 by UEBoy2019

SatishBoddu-MSFT, 2020-10-18:
Hello @UEBoy2019 Could you please accept the same response posted over [the Q&A forum](https://docs.microsoft.com/answers/answers/129465/view.html) ? So that we can close the thread and it will be helpful to other members as well. Thanks for your understanding and for contributing to this page and product.

UEBoy2019, 2020-10-18:
> Hello @UEBoy2019 Could you please accept the same response posted over [the Q&A forum](https://docs.microsoft.com/answers/answers/129465/view.html) ? So that we can close the thread and it will be [...]

OK

UEBoy2019, 2020-10-18:
I have solved the problem successfully

void AzureKinectDevice::k4a2TextureRGB(k4a::image inImg)
{

	uint8_t* data = inImg.get_buffer();
	int T_Width = inImg.get_width_pixels();
	int T_height = inImg.get_height_pixels();
	UTexture2D* NewTexture = UTexture2D::CreateTransient((int32)T_Width, (int32)T_height);
	//if it's size is stable，shouldn't create Utexture2D in while() loop func or tick 
	void* Datas = NewTexture->PlatformData->Mips[0].BulkData.Lock(LOCK_READ_WRITE);
	FMemory::Memcpy(Datas, data, inImg.get_size());
	//call of tick or while(1), FMemory::Memcpy will cause RAM increasing to almost full ,although Temporary object will be released
	//suggest :  FMemory::Memmove(Datas, data, inImg.get_size());
	NewTexture->PlatformData->Mips[0].BulkData.Unlock();
	NewTexture->UpdateResource();
}

Refer to the link
https://github.com/UEBoy2019/Convert-KinectDK-k4a-image-to-Texture2D-for-UnrealEngine-4.25-/blob/master/k4a::image%20to%20Texture2D%20for%20UnrealEngine4.25%20.cpp