microsoft/Azure-Kinect-Sensor-SDK #1385: Water resistance Azure Kinect Device

rapidengineering, 2020-10-16:
In our case water may come near the Azure kinect device. My question is: what is the IP rating of the device.

Also if the device is not water resistant, how to enclose the device in a way to make it water resistant, maybe someone with experience in this field. Obvisously covering up the front with plexiglass will obstruct the Depth camera.

qm13, 2020-10-19:
@rapidengineering Azure Kinect DK is splash proof. It is not water resistant. We have no experience with water tight enclosure for Azure Kinect DK.

rapidengineering, 2020-10-20:
@qm13 Okay, splash proof is not sufficient for our application. So we have to make a water tight enclosure. Which kind of glass/plastic can we use without obstructing the TOV sensor.

karolsenami, 2020-10-22:
Maybe use Google

closed 2020-11-30 by qm13

SimoSR, 2021-02-15:
@qm13 
I have the same issue: i would like to cover the camera with an additional enclosure to increase the protection against dust and water.
Can you provide specifications for glass/plexiglass covering of IR projector and depth sensor to avoid interferences?

strowlands, 2023-08-01:
Just wondering if anyone has a solution to covering the Kinect without obscuring the depth sensor? I am also interested in creating a protective enclosure for the camera.