microsoft/Azure-Kinect-Sensor-SDK #1386: Computer Recommendations

jcova1996, 2020-10-16:
Hi,

I was wondering which computer Microsoft recommends using if I am planning to use 8 Kinects connected to one computer through USB-C to USB-C cables for a body scanning application.

qm13, 2020-10-19:
@jcova1996 we cannot make a recommendation. A number of developers have shared computer specs on this repo. Please search for this on open and closed issues. See #971 as an example.

MaxPalmer-UH, 2020-10-26:
Would be nice to collate different peoples' experiences with computer specs in a single thread. For example in the thread referenced they describe a very high end system that can handle 10 devices. Would be nice to know what performance you might get on a slightly less high end system - e.g. i7 vs. 12 core i9 and how many devices it can handle with a bit of headroom still.

MaxPalmer-UH, 2020-10-26:
Also, I'm really interested to know just what kind of end results people are getting with multi Azure Kinect systems in terms of merged point cloud data and clean-up - i.e. merging point clouds for volumetric capture of people and the tearing you get at boundaries of each point cloud.

MaxPalmer-UH, 2020-10-26:
Finally, there is much knowledge to be collated from end users that doesn't relate to issues and which is more forum based, but nothing really exists. I'll raise an issue that it would be good to have a forum!

MaxPalmer-UH, 2020-10-26:
@jcova1996 how were you planning to use USB C only with so many kinects on a single PC? Is there a solution that you were looking at?

jcova1996, 2020-10-26:
I'm currently using a MackBook Pro with an i9 processor. I'm using a Belkin adapter that converts one USB-C port into 2. My PC has 4 USB-C ports, so with the adapters I have 8. I'm using a long active USB-C cable to connect the Kinects to my setup. This works fine, but there's no way to charge my PC at the same time, so you can only work on limited time. I was also hoping to find a cheaper PC that would still be able to handle all 8 kinects at the same time.

MaxPalmer-UH, 2020-10-26:
So, no problem driving the 8 kinects? What kind of data are you collecting (image, depths, modes, etc?)

closed 2020-11-30 by qm13

Zailux, 2021-02-10:
Any update on this setup ? Does it really work ?
And regarding the USB-C splitter. Are you refering to this product? 
https://www.amazon.de/Belkin-USB-C-Splitter-Ladeadapter-USB-C-Kopfh%C3%B6rer-Adapter-Schnellladen-Schwarz/dp/B086R4KX23/ref=sr_1_3?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=belkin+usb+c+splitter&qid=1612963201&sr=8-3

So you are telling me, u have bought 
4x Splitter USB-C Splitter
8x USB C-to-C active cables (each longer than 1.5m i guess) 

And you had no problems with that setup ?