microsoft/Azure-Kinect-Sensor-SDK #1390: AMD x570 compatibility

madhan001, 2020-10-20:
Is the Azure kinect compatible with the AMD x570 (MSI x570 Tomahawk) chipset's USB 3.1 controller? I don't see any information regarding compatibility with AMD chipsets but only Intel ones.

RoseFlunder, 2020-10-20:
I have tested the azure kinect briefly on a Windows PC with a MSI X570 Carbon Wifi Mainboard.
The k4a viewer worked without issues.
Haven't done any longtime testing though.

madhan001, 2020-10-20:
> I have tested the azure kinect briefly on a Windows PC with a MSI X570 Carbon Wifi Mainboard and AMD Ryzen 3700X.
> The k4a viewer worked without issues.
> Haven't done any longtime testing though.

Was it tested with the back panel port ?

RoseFlunder, 2020-10-20:
Yes, the device was connected via the backpanel usb ports and the pc was running with windows 10.

madhan001, 2020-10-20:
> Yes, the device was connected via the backpanel usb ports and the pc was running with windows 10.

Thanks for answering my questions

closed 2020-10-20 by madhan001