microsoft/Azure-Kinect-Sensor-SDK #1396: What the coordinate system of point cloud is?

ponyzhou404, 2020-10-22:
Hello,

when I'm tried to  acces the center point of a point cloud, which made by:

```
k4a_transformation_depth_image_to_color_camera(transform,img_capture_depth,img_depth2color);
k4a_transformation_depth_image_to_point_cloud(transform, img_depth2color,K4A_CALIBRATION_TYPE_COLOR,img_point_cloud);
```

it shows the xyz as  (-1,-7,535).
Which in my opinion should be (0,0,535).
Is the coordinate system of the point cloud different from the rgb camera?

and another question is
Can I use k4a_calibration_3d_to_2d to project the point cloud back to 2d image coordinate?
I try it like
`k4a_calibration_3d_to_2d(&sensor_calibration,&pc3d_point,K4A_CALIBRATION_TYPE_COLOR,K4A_CALIBRATION_TYPE_COLOR,&pc2d_point,&flag);`
but seems didn't work out.

Thank you.

diablodale, 2020-10-22:
The point cloud origin matches the color camera physical origin due to your parameter of  `K4A_CALIBRATION_TYPE_COLOR`

What do you mean when you write "center point of a point cloud"? There are too many possible definitions of that.

The variance you wrote is a difference of 1 millimeter in the X, and 7 millimeters in the Y, and exact in Z.
It might be possible that the implementation of `k4a_transformation_depth_image_to_color_camera` causes small variations in the upscaling as it creates the mesh. https://docs.microsoft.com/en-us/azure/kinect-dk/use-image-transformation#implementation

This also might be due to the 6 degree down tilt of the depth sensor https://docs.microsoft.com/en-us/azure/kinect-dk/coordinate-systems#depth-and-color-camera. If somehow you had a perfectly centered depth pixel, when that perfect pixel is transformed into the color camera coordinate system, its Y coordinate would be negative due to the tilt, and its X coordinate could also be changed because nothing is perfectly aligned -- the camera calibration matrix used in `k4a_transformation_depth_image_to_color_camera` would adjust these.

zliuchen, 2020-10-24:
My question: function k4a_transformation_depth_image_to_color_camera() takes care of the camera/depth coordination system transform. So, we do not need to specifically write a transform function with the down tilt of Y axis, correct?

For IMU data, the value is returned in the IMU coordinate system, which is usually aligned with world coordinate system. We don't need to transform IMU reading to camera.

diablodale, 2020-10-25:
The documentation I linked above about coordinate systems has good information about both your inquiries:
* Correct. The depth camera is physically tilted down when compared to the color camera. Your example of a depth coordinate that is perfectly centered *on the depth camera* and 535mm away from the *depth camera* will be visually lower to the *color camera*. Therefore the pointcloud coordinates in the *color camera* space for this point will be lower in value than they would have been in the *depth camera* space. You can check this by doing some trigonometry or matrix math. Or keeping everything in depth-space and see if the point cloud for that example point is (0,0,535). Remember that the perfect center of all the cameras is *not* the middle row and middle column of the image. All cameras have distortion and focal centers that are usually not the middle row/column.

You are errant in your understanding of IMU coordinates. The link I provided above defines the IMU coordinate systems and the API provides transformations to move data to/from these coordinate systems. :-)

closed 2020-11-19 by ponyzhou404