microsoft/Azure-Kinect-Sensor-SDK #1398: Pockets of Variation or Invalidation of Depth travel through Depth image

kkh3049, 2020-10-23:
**Bug Description**
While having the Kinect mounted looking down, and viewing the Depth viewer, I will notice every few seconds or so, a Black elliptical form (or multiple) travel through the Depth Camera view.  We've been working with the Kinect since the beginning of the year, and have not seen this before.  What's worse, these artifacts (which I noticed from invalidation assigning the pixel value to zero) will also travel in a depth disparity from the remaining images.  With our application, we need great accuracy and seek to correct Systematic Errors on a per device basis, but we may not be able to if we can't account for these artifacts.
Since noticing them in this way, we've noticed them independent of the orientation of the Kinect as well.

**To Reproduce**
View the Kinect Depth information in the Azure Kinect Viewer from a static position.
(Not sure what circumstances brought on this issue . . . Wondering if we have overused the device?)

**Expected behavior**
Ideally, no odd depth artifacts, travelling through the scene.  At least, only invalidated depths travelling through the scene instead of both invalidated depths and inaccurate depths traveling out through the scene.

**Screenshots**
This GIF shows one such artifact traveling through the scene.
[image]

**Desktop (please complete the following information):**
 - OS with Version: Windows 19041
 - SDK Version:  1.4.0
 - Firmware version:

diablodale, 2020-10-26:
As an observation...I call these "amoebae" given their round-like shapes and organic movement. I see them with Kinect v1, v2, and Azure. Now 8+ years with them and across hardware/technology, I have considered them part of the tech...a known aberration I can usually filter out.

I'm not suggesting there isn't a bug. Rather, I guess it is something systemic in the tech or methodology given this exists across all generations of Kinect and having almost identical shape and movement. Maybe it is a particle of dust floating close to the emitter or sensor. 🌠 Or a fairy 🧚‍♀️😉

qm13, 2020-10-26:
This is not a bug. The moving blob is dust in the environment.

kkh3049, 2020-10-29:
Huh.  Dust?  I didn't even think of that. I'm not sure I have a way to validate that here, but that could make sense; we recently had construction in our office building, so maybe that's why we're noticing so much more now as opposed to a month or so ago.
That's unfortunate, but we'll work around it.

closed 2020-10-29 by kkh3049