microsoft/Azure-Kinect-Sensor-SDK #1399: what version of Visual Studio to use, for both Sensor SDK and body tracking SDK?

zliuchen, 2020-10-24:
Hi, there,
    From this page:
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md
    It is mentioned that we shall use **VS2017** in order to build and install the sensor SDK.

   However, from this page:
https://docs.microsoft.com/en-us/azure/kinect-dk/body-sdk-setup
    It is recommended we install the **VS redistributable for VS 2015**. 
    What is this for, for the Depth engine or Nvdia card?

   So, If I want to build the source for both Sensor SDK and body tracking SDK, what version of VS to use?

    If I choose to use MSI, this means I don't have the ability to compile but can use the library directly. In this way, I can use either VS 2015 or VS 2019, correct?

Thanks.

diablodale, 2020-10-25:
There is a distinction.
1. To compile the SDK itself
2. To compile your own applications using the SDKs (core sensor and body tracking)

The first link you provided is to compile the core Kinect SDK itself. While you can technically do it, it is a very rare activity. I recommend you follow the instructions given at that link.

The second link you provided is to compile your own applications using the body tracking SDK. The Body Tracking DLL and libraries need the VS redistrib. Luckily the VS redistrib has now been unified and this unified VS redistrib for 2015-2019 is often automatically installed on Windows 10 via Windows Update. You can also download this unified VS redistrib for 2015-2019 from many locations like https://visualstudio.microsoft.com/downloads/

It is not possible to compile the body tracking SDK itself. It is closed-source.

zliuchen, 2020-10-25:
Thank you for your answers.  Just to confirm, below is what I can do:

1. download MSI for both Kinect SDK and body tracking SDK
2. download the VS redistrib as suggested by the page, for VS 2015
3. Use VS 2019, and the 2015 redistrib is compatible with VS 2015.

Is this correct? Thanks.

diablodale, 2020-10-26:
1. Yes. These two MSI's will give you the DLLs and libraries needed to compile your own applications.
2. Yes. You will need the VS redistrib to run your own applications that are compiled/linked with (1). You can compile without it, but when its time to run, you need the redistrib.
3. I'm a bit unclear what you are asking. The runtime redist and the ABI are specific and versioned. You do have to follow the rules. https://docs.microsoft.com/en-us/cpp/porting/binary-compat-2015-2017?view=vs-2019

closed 2020-11-30 by qm13