microsoft/Azure-Kinect-Sensor-SDK #1401: Any PCIe USB cards with powered USB C and dedicated controllers per port

MaxPalmer-UH, 2020-10-26:
You can find PCIe USB 3 cards that have one dedicated controller per point, but these all have a USB A connector. Are there any that have a powered UCB C connector with one controller per port?

qm13, 2020-10-26:
Are you asking for PCIe cards that support USB 3.1 Gen 1 power delivery?

MaxPalmer-UH, 2020-10-26:
Yes, I guess. My understanding is that USB C ports are able to deliver power and data to the Kinect, but that these ports need to meet the power requirements (which presumably most don't). The advantage of that is that you only need one cable from the host to the Kinect.  However, I haven't seen expansion cards (with dedicated controllers) with multiple powered USB C ports on them. They are usually a mix with mainly USB A that can't deliver the power. I presume there aren't USB A cards that can meet the power requirements and you have to provide power separately?

tim-depthkit, 2020-10-28:
I have not tested this card, but it looks promising: https://www.delock.com/produkt/90308/merkmale.html

DrMaxP, 2020-10-28:
Thanks, this looks really good.

closed 2020-11-30 by qm13

nick-cr, 2020-12-04:
@tim-depthkit did you get this card in the end? Did it work?

tim-depthkit, 2020-12-04:
@nick-cr I have not yet, but one of our customers has one is is going to test it. I'll post back here once I hear about the results.

MaxPalmer-UH, 2020-12-07:
Looking forward to hearing how this goes Tim

caseyfarina, 2021-04-05:
@tim-depthkit Any update on that card? Thanks!

guss2149, 2021-05-19:
I have not tried that one. It's too long of a lead time for us here in the US. But I've tried the Startech https://www.startech.com/en-us/cards-adapters/pexusb312c3 which does not work. I called Startech and he helped me to see what chipset is compatible with the Azure Kinect. From the product page: "For the Azure Kinect DK on Windows, Intel, Texas Instruments (TI), and Renesas are the only host controllers that are supported."

I'm trying to find a card that can handle two Kinect DK cameras plugged in via active USB-C cables 6m long. This is the cable we are trying: https://www.ntcdistributing.com/usb-repeater/active-cable/usb-c-active-cable-for-azure-kinect-with-pd-6m-10m/

Has anyone had any success with any USB-C PCIe cards available fairly quickly in the US?

DrMaxP, 2022-09-04:
Any updates on this? Anyone found USB-C cards which can supply power and data that work? Ideally 4 ports or above per card with dedicated controllers per port. Needs the correct USB chipset...