microsoft/Azure-Kinect-Sensor-SDK #1402: Any suggestions on tools to debug memory leaks?

jackjansen, 2020-10-27:
Do you have any suggestions on a tool to use to find memory leaks? I presume the memory leaks are in my (multi-threaded) code, not in the SDK, but because all the allocations are done by the SDK using a compile-in leak detector on Windows doesn't work.

On MacOS there is a nifty `leaks` tools that would work (because it overrides all allocations, also from dynamic libraries, by preloading a malloc replacement) but the SDK doesn't work on MacOS.

I have run my program on Linux and used `valgrind`, and that could in principle do the job, but it slows down the capturing process so much that I just get a slew of failed captures and frames dropped in the grabber.

diablodale, 2020-11-02:
I'll volunteer my perspective. On Windows and the Microsoft compiler, I use https://docs.microsoft.com/en-us/visualstudio/debugger/finding-memory-leaks-using-the-crt-library?view=vs-2019. Since it will be a debug build, no optimizations will occur...therefore high CPU cost. As Heisenberg or Schrödinger might say, there is an effect when making a measurement. 😹

When I do such tests, dropping frames doesn't concern me since I'm not testing that area...I'm testing for memory leaks. Plus, it is expected even in Release builds for some frames to drop. Therefore, the test scenario isn't invalidated. One area that could differ is the optimizations. It is technically possible for the optimizations/not of Release and Debug builds to lead to false-positives or false-negatives.

Perhaps you can explore addresssanitizer via window clang support https://devblogs.microsoft.com/cppblog/addresssanitizer-asan-for-windows-with-msvc/

FYI the Kinect SDK provides an interface for you to own allocation https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1274

Last, I use tools at the coding stage (e.g. C++ and RAII) to prevent memory leaks before they can occur. I believe that in 2020, no coder using the SDK should be writing bare malloc(). Such bare usage is at a lower level (the SDK itself or kernel drivers).

jackjansen, 2020-11-03:
Thank you for the suggestion, @diablodale, but if I read the MS docs I get the impression that it will _only_ find leaks in the code that you have compiled with this library, _not_ leaks of memory that has been allocated by a DLL you've loaded (such as, in my case, the K4A DLL).

#1274 is going to be interesting, once implemented, because that would allow me to do my own allocator implementation (and therefore I could use the MS memory debugger).

(and in the mean time I have found my original problem using valgrind: if my captures were not forwarded to the next step in my multithreaded pipeline I could lose a reference to the depth image in the capture. But only if the next thread in the pipeline was too busy to pick up the capture in time).

diablodale, 2020-11-03:
I think there is an opportunity to increase coverage using those tools today.

1. Write your own k4a_memory_allocate_cb_t allocator. That works today. When you get image buffers they will be aligned on 32. The issue #1274 I linked is focused on allowing alignment of the images buffers to be selectable rather than the forced 32.
2. Compile the K4A SDK itself. It is open source at this git repo. Only a subset, the depthengine, is closed. As is all of body tracking.
3. Yes, you do have to compile with address sanitizer. However, it does allow you to compile an optimized Release build. Which is an improvement over the VSCRT approach.

Step 1 alone increases your coverage using the VSCRT memory tracking or address sanitizer. All three together would expand coverage across your app/dll in a Release build except for depthengine, the c/c++ runtimes, and any other code/DLLs you might link. I'm kinda curious how well this address sanitizer approach would work. For future use. 😉

jmachowinski, 2020-11-07:
On linux try heaptrack (https://github.com/KDE/heaptrack)
Its fast, and should point you to the point were the memory 
gets allocated.

closed 2020-11-30 by qm13