microsoft/Azure-Kinect-Sensor-SDK #1403: Linux libdepthengine.so.2.0

nrpr161293, 2020-10-27:
**Describe the bug**
My team is trying to run the k4aviewer on linux machine and the depth info of the kinect isn't working, other info like RGB is working perfectly. I think the error is related with the libdepthengine.so.2.0 but we already put the file in the build/bin folder and still for some reason we cannot read it. Can you please help us?

**To Reproduce**
1. open the k4aviewer on terminal
2. select "enable depth camera"

**Expected behavior**
It should work with depth info.

**Logs**

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS with Version: Linux 18.04
 - SDK Version:  1.4.1
 - Firmware version:  
[image]

RoseFlunder, 2020-10-27:
Did you compile the sdk yourself?
Usually you don't need to copy any files when installing the sdk and k4aviewer via apt from the MS package repository.

nrpr161293, 2020-10-28:
Yes we compile using this instructions https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md, but doesn't generate any libdepthengine.so.2.0 on the build/bin folder, for that reason we copy the lib to the folder.

zliuchen, 2020-10-29:
I have similar issue on Windows. I built from cmake, and the depthengine dll is not part of the project's build.lib folder. Therefore, I copied the depth engine dll from the msi installer, but it didn't help. Still complained about missing depth engine

qm13, 2020-10-29:
> Yes we compile using this instructions https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md, but doesn't generate any libdepthengine.so.2.0 on the build/bin folder, for that reason we copy the lib to the folder.

The depth engine is provided as a binary only.

qm13, 2020-10-29:
Did you install the [Debian package](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md#debian-package) or build the SDK from scratch? If you built from scratch please try installing the Debian package and run k4aviewer.

nrpr161293, 2020-10-29:
Hi @qm13 thank you for your feedback regarding to the depthengine. Indeed is a binary and for that reason it cannot be included on the sdk. For that reason I copy it and paste it on the build/bin folder, I don't know if this should be the correct one?
Regarding to the installation, we try both, first using the debian package and then build from scratch, in both ways we can open the k4aviewer and the RGB image works perfectly but when we try to see the depth info the following errors appear. Maybe I'm putting the libdepthengine.so.2.0 in the wrong folder.

[image]

qm13, 2020-11-30:
What happens when you use the pre-built k4aviewer tool in the package?

closed 2021-01-08 by qm13

jiapei-nexera, 2022-07-19:
I'm now having exactly the same problem.... Where can we download that **libdepthengine.so.2** ?