microsoft/Azure-Kinect-Sensor-SDK #1406: fitting the pose of a body model to the result the poses of the kinect joints

KiyarashFarivar, 2020-10-29:
**Describe the question**
I want to set the pose of a skeleton model (similar to the one Azure kinect uses) to the pose of the body joints given by the sdk. 
(I have read the Joint documentation (https://docs.microsoft.com/en-us/azure/kinect-dk/body-joints) but I'm still confused.)

my question are: 

0. What does it mean that the joint orientations are relative to the depth sensor coordinate ? does it mean that if I apply the rotation to the depth coordinate system I get the joint coordinate system ? what does a joint coordinate system tell me about the joint ?

1. how should I apply the quaternion rotations ? 

- Let's consider the pelvis joint. What skeleton pose should I start from and then apply the pelvis quaternion rotation to, to get the both bodies facing the same direction ? 

- Also , to apply the rotations do I have to center the whole body around the joint and then apply the rotation ? (I guess it kind of relates to the previous question)

2. In what order should I apply the rotations to my skeleton ? should I start from parent to child in the kinect's joints table? 

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS with Version: Win 10
 - SDK Version:  1.4.1

**Additional context**
The body model I am interested in is the SMPL model but I only use the pose parameters. Although I don't think this would be relevant to the question.
I know the joints are not the same (and they have different mechanics) and I will eventually have to interpolate, But even if the joints where the same that still wouldn't answer the questions above.

knat, 2020-10-30:
https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples/sample_unity_bodytracking doesn't answer your questions, maybe it can give you some hints. 3D quaternion is hard for most of us :)

closed 2021-01-08 by qm13

liuyuex97, 2021-12-07:
@kfarivar Have you found out the transformation between kinect and smpl's axis-angle representation? I'm facing the same problem and couldn't find the answers from documentations.