microsoft/Azure-Kinect-Sensor-SDK #1407: tracker::pop_result() low performance

aspioupiou, 2020-10-29:
Hi all, 

This is more of a question than a bug report.

We integrated the azure kinect SDK for headtracking purpose. The integration works like a charm with our company tools.

However, we spotted a huge performance drop when connecting our libraries with 3rd party softwares. 
After some benchmark, we spotted the bottleneck: `ka4bt::tracker::pop_result();` is the culprit.
It seems this call freezes for some time the application and causes the FPS drop.

- Playing with the `timeout` parameter had no effect
- Calling this function inside a dedicated thread had no effect.
- GPU and CPU are not overloaded.

Am I missing something??

Cheers,

qm13, 2020-10-29:
Without knowing anything about your company tools we are just guessing at the issue. One area to look at is memory transfer between the GPU and CPU.

aspioupiou, 2020-10-29:
Hi @qm13 
Thanks for your reply !

Indeed, it might be difficult for you to guess what happens without knowing the setup.
I'm just asking for some advice or if something is wrong in the way I handle the kinect .

Datas transfer could be an issue, but neither the CPU nor the GPU are overloaded (25% and 80% respectively) The GPU bandwith usage is also in the usual limit. 

Could it be related to the OpenGL context? What happens if I start the Kinect on the main thread and then I execute only some code on a dedicated thread? Is there any autonomatic context switch?

What's the data worflow of the kinect ? Is it  Kinect -> CPU -> GPU ?

qm13, 2020-11-02:
@aspioupiou the perf tools especially GPU tools often do not take into account DMA orchestrated data transfers.

The data flow is Kinect -> CPU -> GPU

closed 2020-11-30 by qm13