microsoft/Azure-Kinect-Sensor-SDK #1409: Azure Kinect compatibility with USB Host Controllers

SimoSR, 2020-11-01:
**Describe the bug**

I'm searching for a PCIe expansion card for my intel motherboard to make available an additional type C port to plug the AKDK camera via a single type C cable (without power supply connector).

I know the issue compatibility of AKDK SDK with ASMedia USB host controllers, but i'm currently not able to find a USB type C PCIe expansion card with a chipset from a different manufacturer (for example Intel, Texas Instruments, Renesas).

This is the card I would like to try:
https://www.startech.com/en-gb/cards-adapters/pexusb312c3
The chipset is ASMedia ASM3142.

Can someone confirm the compatibility of this card with the AKDK SDK?
Is there any alternative way to go?

jmachowinski, 2020-11-07:
Linux or windows ?

In general, we found out that the RGB Camera is causing the issues,
as long, as you only activate the depth sensor, it should work.

SimoSR, 2020-11-07:
We are working on win10 and we need both RGB and depth stream.

jmachowinski, 2020-11-07:
We are only using linux, so I can't help you with the windows stuff.

oferoze, 2020-12-04:
We are having the same issue. Have tried the Allegro Pro 4 USB 3.2 Gen 2 Type-C PCIe card with the ASM3412 chip set. Depth by itself works fine, but the RGB camera is not working. We have been using it with these cables https://www.newnex.com/azure-kinect-extension-cable.php.

I have been able to use the cable successfully from my laptop which is a dell xps 15 

We are using ubuntu 18.04

SimoSR, 2021-01-31:
Any news regarding compatibility with ASMedia Controllers? at the moment it seems that chipset is the only one adopted on PCIe cards with USB type-C ports available on the market.

ageorgop, 2021-03-15:
I'm also interested in getting these chips to work.  I would even gift someone at Microsoft a card to make this happen faster.