microsoft/Azure-Kinect-Sensor-SDK #1410: Kinect Azure trying to connect to Silex DS-600 (and or DS-510)

Karagorion, 2020-11-01:
Dear All, 

I kindly wanted to ask the community for help in a connection problem with the Kinect Azure.

I want to achieve a setup where I can connect 1 or 2 Kinect Azure over 30-50 Meteres to my Laptop.

Therefore I bought for tests a Silex DS-510 and a Silex DS 600.

This decive gives me the ability to connect usb2 and usb3 Devices through my Ethernet Port.
In my first try I was kinda happy that the Kinect Azure got Connected without a problem by the Silex SX Virtual Link.
---> https://gyazo.com/ceca7a6f323dcb3ea959a18a8dcb9092

The problem is (probably it has something to do with USB Ports) that my Kinect Azure appears in the Device Manager as a Device that has no Drivers.
---> https://gyazo.com/0a77598b7d9b458032eeebd83aa42cc3

The Kinect Azure also could not be found in the Azure Kinect Viewer.

Is my workflow right that I am trying out or would you connect Kinect Azure over big distance with different devices?

Thank you very much in advance for your help.

With kind regards
Karagorion

wes-b, 2020-11-02:
Have seen #427 about USB extenders?

Karagorion, 2020-11-15:
Thank you very much for your help :) wes-b :)

jmachowinski, 2020-11-18:
Hi,
we are using IDS Fiber optics USB Extender in our setup.
https://de.ids-imaging.com/store/products/accessory/interface-cables/usb-3-0-aoc-active-cable-straight-screwable-50-m.html

They cause some trouble, with power on / rebooting but work in general.
We worked around these issues, by adding an external relay, to power up the azures,
after the USB extender was powered and the pc was booted.

Note, we use linux only, therefore I can't say if it will work under windows.

closed 2020-11-30 by qm13