microsoft/Azure-Kinect-Sensor-SDK #1412: Color sensor not grabbing an image

SinisterMJ, 2020-11-10:
I encountered the precisely same issue as in #1237, but I need to get it to run reliably on my or other systems, or at least understand what causes it. 

I have described the issue, with my code, at https://stackoverflow.com/questions/64770775/camera-not-grabbing-a-color-image

If I compile the SDK myself, in the k4aviewer.exe I get the depth and ir image shown just fine, the camera image though does not show.

In the pre-compiled installer, the k4aviewer shows all images, depth, ir and color, correctly. I was not able to figure out what causes this.

qm13, 2020-11-12:
Based on your comments both here and stackoverflow it appears you have some sort of build issue. Can you please verify you correctly followed the build instructions [here](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md)?

SinisterMJ, 2020-11-13:
Originally I built it via a conan package we created. I just now created the file exactly according to the documentation, same behaviour: I can grab the IR and depth image, the color image does not work. I tested with the k4aviewer.exe to verify that its not my code causing this.

SinisterMJ, 2020-11-13:
> 
> 
> Originally I built it via a conan package we created. I just now created the file exactly according to the documentation, same behaviour: I can grab the IR and depth image, the color image does not work. I tested with the k4aviewer.exe to verify that its not my code causing this.

Edit;

Toolchain used:
cmake version 3.19.0-rc1
ninja 1.10.1
Visual Studio 2017 Community 15.9.29

SinisterMJ, 2020-11-18:
I was at a colleagues system today, he has the precisely same hardware configuration. We checked on his system, he also received no color image. After some further testing we checked whether some of our company policy installs could be causing this, and when we disabled Kaspersky Virus Scanning, it worked. Enable Kaspersky, no color image again. 

No idea why this would happen only with self-compiled libraries, but that solves this issue.

closed 2020-11-19 by SinisterMJ