microsoft/Azure-Kinect-Sensor-SDK #1413: Body tracking converted to screen coordinates

TomitaTseng, 2020-11-11:
I'm sorry to bother you.

I use C# for program development.
I want to convert body tracking to screen coordinates.

How to draw a point on the screen after getting the position coordinates of the Head joint.

Could you please tell me, thank you very much.

qm13, 2020-11-12:
Look at the body tracking viewer sample code [here](https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples/simple_3d_viewer).

TomitaTseng, 2020-11-12:
Could you please tell me, thank you very much.
I use C# for program development.
Are there any C# examples?

RoseFlunder, 2020-11-12:
Just look into the API Reference:
[TransformTo2D](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/release/1.4.x/struct_microsoft_1_1_azure_1_1_kinect_1_1_sensor_1_1_calibration_a31828fb182a5386b6ca6c72e4c08dea6.html#a31828fb182a5386b6ca6c72e4c08dea6)

TomitaTseng, 2020-11-16:
> Just look into the API Reference:
> [TransformTo2D]

Thank you for your guidance.
I tested TransformTo2D to convert the screen's x and y coordinates, but the coordinate error is large.
Is there a C# program example for TransformTo2D?
Could you please tell me, thank you very much.