microsoft/Azure-Kinect-Sensor-SDK #1415: is there any point cloud viewer sample code?

bjkoo88, 2020-11-13:
I am currently building a real-time point cloud viewer with Azure Kinect Sensor. 
Since I am new to this world, I am having trouble to build it.

The viewer has to show the point cloud data(with colors) within a certain depth range just like the output of the transformation example. 

Mr. Takashi Yoshinaga's  program shown in the link below is something I want but with one Azure Kinect Sensor.
(https://www.youtube.com/watch?v=Y45RJYq_zuk)
Is there any sample code of such program that I can use?

Any help will be welcomed!

RoseFlunder, 2020-11-13:
The official viewer that comes with sdk itself is opensource in this repository.

bjkoo88, 2020-11-13:
> The official viewer that comes with sdk itself is opensource in this repository.
> 
> https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/tools/k4aviewer

Yes, I have tried to use them but the thing is whenever I try to build the official viewer with visual studio after all the CMAKE process is done, it gives me some errors. 
At first, it gives me LNK1181 error saying cannot open k4arecord.lib and I solved it.
Then it gives me this error which is also LNK1181 error 
[image]

For this I couldn't solve the problem so I am stuck.
I am afraid that I am doing something wrong here or from the beginning?

qm13, 2020-11-19:
Have a look at https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/fastpointcloud

bjkoo88, 2020-11-19:
> Have a look at https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/fastpointcloud

I am looking at this fastpointcloud sample. What can I get from this example?

qm13, 2020-11-30:
@bjkoo88 this is an example of generating a point cloud. The k4aviewer source also has a sample.

closed 2021-01-08 by qm13