microsoft/Azure-Kinect-Sensor-SDK #1417: Unable to find Container ID

henryo12, 2020-11-14:
**Describe the bug**
An error is thrown when trying to open a Kinect device. It appears to be a container ID issue. I have 3 Kinects, but this error only occurs when opening one specific devices. The other two function properly.

**To Reproduce**
1. Plug in affected Kinect
2. Open the Azure Kinect Viewer
3. Click Refresh Devices

**Expected behavior**
The affected camera should be available and able to open. It does open properly on another PC.

**Logs**
This is the output in the Azure Kinect Viewer when I Refresh Devices:
```
[ trace    ] : k4a_device_t_create(). Created   k4a_device_t 000001C21DD3ED10
[ trace    ] : depthmcu_t_create(). Created   depthmcu_t 000001C21DD29B30
[ trace    ] : usbcmd_t_create(). Created   usbcmd_t 000001C216B54020
[ info     ] : find_libusb_device(). Container ID found: {abe70aee-af42-4fea-d086-68c9fd85aeed} 
[ info     ] : populate_serialnumber(). Serial Number found 000141193812
[ trace    ] : usb_cmd_io(). XFR: Cmd=00000115, PayloadSize=255
[ trace    ] : colormcu_t_create(). Created   colormcu_t 000001C21DDAFDF0
[ trace    ] : usbcmd_t_create(). Created   usbcmd_t 000001C216B541E0
[ error    ] : find_libusb_device(). Unable to find Container ID: {abe70aee-af42-4fea-d086-68c9fd85aeed} 
[ trace    ] : usbcmd_t_destroy(). Destroyed usbcmd_t 000001C216B541E0
[ error    ] : usb_cmd_create(USB_DEVICE_COLOR_IMU_PROCESSOR, NULL_INDEX, container_id, &colormcu->usb_cmd) returned failure in colormcu_create()
[ trace    ] : colormcu_t_destroy(). Destroyed colormcu_t 000001C21DDAFDF0
[ error    ] : colormcu_create(container_id, &device->colormcu) returned failure in k4a_device_open()
[ trace    ] : usbcmd_t_destroy(). Destroyed usbcmd_t 000001C216B54020
[ trace    ] : depthmcu_t_destroy(). Destroyed depthmcu_t 000001C21DD29B30
[ trace    ] : k4a_device_t_destroy(). Destroyed k4a_device_t 000001C21DD3ED10
```

**Screenshots**
This is the device listed in the Device Manager:
[image]

**Desktop (please complete the following information):**
 - OS with Version: Windows 10 Edu
 - SDK Version:  1.4.1
 - Firmware version: 
  RGB camera firmware:      1.6.110
  Depth camera firmware:    1.6.79
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft

**Additional context**
The issue started happening after I tested the Kinect with an [active USB cable](https://www.amazon.com/gp/product/B00DMFFL2W/ref=ox_sc_act_title_3?smid=A1AMUYYA3CT6HJ&psc=1). I also recently installed a [USB PCIe card adapter](https://www.startech.com/en-us/cards-adapters/pexusb3s44v), so the issue could have come from either of those two changes.

I tried to uninstall the device from the device manager and restarted my PC, but the issue persists. I also tried resetting the device with the AzureKinectFirmwareTool, but the reset fails due to the container ID issue. Thanks for your help.

henryo12, 2020-11-18:
I reinstalled Windows, and that fixed the issue. It would be helpful to hear another another potential solution in case this happens again. Otherwise, we can close the issue.

qm13, 2020-11-20:
@henryo12 you mention you fixed this with a Windows reinstall. Are you now using it with the USB cables that came with the camera or are you now using it with the active cable?

henryo12, 2020-11-20:
I am now able to use with both the stock cable and the active cable, whereas the camera would not work with any cable previously. 

The original active cable I tested with turned out to be broken, but I had ordered three of those cables and the other two worked. Maybe that broken cable corrupted something on my computer, I’m not sure.

closed 2020-11-30 by qm13

VitaliyKurokhtin, 2020-12-09:
Two things to note:
- The difference in container id in logs is due to: 1438
- Same error I had I was able to resolve by removing hidden Azure Kinect 4K Camera (and all related devices) from Device manager. To do so, open device manager, select View\Show hidden devices and look for anything Kinect related that has translucent icon. 1439

henryo12, 2020-12-09:
Thanks @VitaliyKurokhtin , that looks exactly right.

natelowry, 2022-04-16:
In case these instructions don't work for folks, I had to uninstall the hubs one level up and restart to resolve this issue.  Both the `Generic SuperSpeed USB Hub` (containing the cameras) AND `Generic USB Hub` (containing the Microphone)

[image]

underbelly69, 2023-04-13:
yep - uninstalling all the hidden usb things and then a reset of the device worked for me

ofTheo, 2023-09-11:
@VitaliyKurokhtin and all curious why the #1438 bug not fixed by now? 
it sounds like it should be a straightforward fix. 

is there a patch anyone has for this issue that can be applied?

thanks!
Theo