microsoft/Azure-Kinect-Sensor-SDK #1418: Is polling the only way to capture frames?

oafolabi, 2020-11-15:
Hi, 

I am working on an application that grabs frames and saves them to file.  I am new with this device, but so far the only way I can grab frames is through polling. I am a bit concerned that my application might drop some frames if the time between polling is too long. I would prefer to use a callback if available. Is there a way to grab frames using a callback function?

Thanks

RoseFlunder, 2020-11-15:
No there isn't.
Maybe you could work with threads though so that the polling thread isn't busy with processing the frames.

Chris45215, 2020-11-16:
First off, the SDK includes a Recorder which saves the frames, so I hope you aren't writing that code yourself. The guide to use it is at https://docs.microsoft.com/en-us/azure/kinect-dk/record-sensor-streams-file and with more detail at https://docs.microsoft.com/en-us/azure/kinect-dk/azure-kinect-recorder. I don't remember if that Recorder code is an open part of the SDK, but if so it's worth looking for and using it as much as possible rather than writing your own code from scratch.

If you have a SSD then there shouldn't be a problem saving frames, and most HDDs should be able to handle it too - I've saved the full color and depth datasets from the sensors onto my computer for 30-second periods, and they are about 3 GB. So, the data rate is about 100MB per second. You can economize slightly by ignoring the audio and gyroscope data, or even more by getting only the RGB or only the depth images if you only want one. And you can specify both the resolution and framerates of the image frames, so that can be used to reduce the data amount by a lot. 

Also, each frame includes a timestamp, and that can give a good indication if any frames are dropped. The intervals are not exactly perfect of course, but they will make it clear if a frame is missing.

oafolabi, 2020-11-16:
@RoseFlunder  and @Chris45215 , thank you for your quick responses. I took a look at the recorder, and it almost meets my needs. I would also like to visualize the frames (and maybe process) them as I am recording. The idea is that this will give me some feedback on what area of space I should be recording and how fast I should be moving the camera. Ideally, I would want to combine something like [kinectfusion](https://github.com/microsoft/Azure-Kinect-Samples/blob/master/opencv-kinfu-samples/README.md) with recording. Do you have any suggestions for this? Alternatively, I can run the recording application and also write some code to grab frames and visualize them in a different program, but I doubt this will go well.

Chris45215, 2020-11-16:
For visualizing the frames, without recording, I use the bundled Azure
Kinect Viewer app and see how things look in real time, though I don't
think its code is in the samples (and it will claim exclusive use of the
sensor when it is running). The Azure Kinect Samples / Body Tracking
Samples folder does have a lot of samples that visualize the depth data,
and you can remove the body-tracking-specific code from those. That will
only show the image the camera is currently seeing, and you can add your
own code to save the frames or do whatever processing is needed.

I hope I'm understanding your request correctly.

oafolabi, 2020-11-17:
@Chris45215 , thanks. I'll take a look at the body tracking code. Do you know if the Recorder also claims exclusive use of the sensor?

Chris45215, 2020-11-17:
I'm not certain, but it probably does - in my applications, only one
program can pull data from the sensor at a time (I think that's true for
all USB-connected cameras in general). But there's nothing preventing you
from writing one application that uses it in multiple ways within that
program. And you can always have one program output the sensor's info to
other programs.

oafolabi, 2020-11-17:
I see, thanks. This has been very helpful. I will close this issue for now.

closed 2020-11-17 by oafolabi