microsoft/Azure-Kinect-Sensor-SDK #1419: Impossible to acquire image from rgb camera from k4aviewer

jacquesj1, 2020-11-18:
Hello,
When I execute k4aviewer, I have the following message :
[ warning  ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c:285 ) : depth_engine_thread(). Dropping depth image due to bad timestamp at startup
[ error    ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c:144 ) : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:     17733 type:Depth
[ error    ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c:144 ) : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:     51066 type:Depth
[ error    ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c:144 ) : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:     84400 type:Depth
[ error    ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c:144 ) : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    117733 type:Depth
[ error    ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c:144 ) : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    151066 type:Depth
[ error    ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c:144 ) : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    184400 type:Depth
[ error    ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c:144 ) : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    217733 type:Depth
[ error    ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c:144 ) : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    251066 type:Depth
[ error    ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c:144 ) : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    284400 type:Depth
[ error    ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c:144 ) : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    317733 type:Depth
[ error    ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c:144 ) : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    351066 type:Depth
[ error    ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c:144 ) : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    384400 type:Depth
[ error    ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c:144 ) : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    417733 type:Depth
[ error    ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c:144 ) : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    451066 type:Depth
[ error    ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c:144 ) : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    484400 type:Depth
[ error    ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c:144 ) : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    517733 type:Depth
[ error    ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c:144 ) : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    551066 type:Depth
[ error    ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c:144 ) : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    584400 type:Depth
[ error    ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c:144 ) : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    617733 type:Depth
[ error    ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c:144 ) : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    651066 type:Depth
[ error    ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c:144 ) : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    684400 type:Depth
[ error    ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c:144 ) : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    717733 type:Depth
[ error    ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c:144 ) : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    751066 type:Depth
[ error    ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c:144 ) : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    784400 type:Depth
[ error    ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c:144 ) : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    817733 type:Depth
[ error    ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c:269 ) : capturesync_get_capture(device->capturesync, capture_handle, timeout_in_ms) returned failure in k4a_device_get_capture()
[ error    ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c:359 ) : cmd_status == CMD_STATUS_PASS returned failure in depthmcu_depth_stop_streaming()
[ error    ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c:362 ) : depthmcu_depth_stop_streaming(). ERROR: cmd_status=0x00000063

Windows concerning IR, depth and rgb don't appear.

When I disabled the RGB camera and restart acquisition (Depth camera enable), I can see all acquisitions and have the following messages:
[ warning  ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c:278 ) : depth_engine_thread(). Depth image processing is too slow at 1000ms (this may be transient).
[ warning  ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c:278 ) : depth_engine_thread(). Depth image processing is too slow at 1000ms (this may be transient).
[ warning  ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c:278 ) : depth_engine_thread(). Depth image processing is too slow at 1000ms (this may be transient).
[ warning  ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c:278 ) : depth_engine_thread(). Depth image processing is too slow at 1000ms (this may be transient).
[ warning  ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c:278 ) : depth_engine_thread(). Depth image processing is too slow at 1000ms (this may be transient).
[ warning  ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c:278 ) : depth_engine_thread(). Depth image processing is too slow at 1000ms (this may be transient).
[ warning  ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c:278 ) : depth_engine_thread(). Depth image processing is too slow at 1000ms (this may be transient).
[ warning  ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c:278 ) : depth_engine_thread(). Depth image processing is too slow at 1000ms (this may be transient).
[ warning  ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c:278 ) : depth_engine_thread(). Depth image processing is too slow at 1000ms (this may be transient).
[ warning  ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c:278 ) : depth_engine_thread(). Depth image processing is too slow at 1000ms (this may be transient).
[ warning  ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c:278 ) : depth_engine_thread(). Depth image processing is too slow at 1000ms (this may be transient).
[ warning  ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c:278 ) : depth_engine_thread(). Depth image processing is too slow at 1000ms (this may be transient).
[ warning  ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c:278 ) : depth_engine_thread(). Depth image processing is too slow at 1000ms (this may be transient).
[ warning  ] ( /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c:278 ) : depth_engine_thread(). Depth image processing is too slow at 1000ms (this may be transient).
[... cut ...]
[ 3218.199859] xhci_hcd 0000:04:00.0: Looking for event-dma 00000000350c2080 trb-start 0000000035181b70 trb-end 0000000035181b70 seg-start 0000000035181000 seg-end 0000000035181ff0
[ 3218.199859] xhci_hcd 0000:04:00.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 2 comp_code 13
[ 3218.199860] xhci_hcd 0000:04:00.0: Looking for event-dma 00000000350c2090 trb-start 0000000035181b70 trb-end 0000000035181b70 seg-start 0000000035181000 seg-end 0000000035181ff0

**Desktop (please complete the following information):**
 - OS with Version: ubuntu18.04
 - SDK Version:  1.3
 - Firmware version:  1.6.102 for rgb cam and 1.6.75 for depth cam

jacquesj1, 2020-11-18:
I try my azure kinect on windows 10 ==> no problem !!

qm13, 2020-11-19:
Please upgrade to the latest SDK and firmware versions. Please submit a fresh log if you are still experiencing problems.

jacquesj1, 2020-11-19:
Hello,
If i  update, can I use body sdk on Ubuntu 18.04.?
Best regards.

qm13, 2020-11-30:
Yes, you can use Body Tracking SDK on Ubuntu 18.04

closed 2020-11-30 by qm13