microsoft/Azure-Kinect-Sensor-SDK #1424: RGB-D alignment between color and depth with CUDA

Mustardsauce, 2020-11-24:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

**Describe alternatives you've considered**

I did the alignment between color and depth image(Azure kinect dk) with CUDA. It takes so long time because it is necessary to tranform the 2D-3D conversion. I've done improvement for the alignment through CUDA.
https://github.com/Mustardsauce/Azure-kinect-alignment-with-CUDA

[image]

**Additional context**

qm13, 2020-12-01:
Nice work @Mustardsauce

UnaNancyOwen, 2020-12-02:
Hi @Mustardsauce, Thank you for sharing! Good job! 👍
I think it would be nice to add code to measure the respective processing time of CPU and GPU to make it easier to compare.
It is so important to reproduce on other emvironments. I expect that.

Mustardsauce, 2020-12-02:
> Hi @Mustardsauce, Thank you for sharing! Good job! 👍
> I think it would be nice to add code to measure the respective processing time of CPU and GPU to make it easier to compare.
> It is so important to reproduce on other emvironments. I expect that.

Thank you for your comment!

I agree with your opinions.
I'll add the code (to measure the respective processing time of CPU and GPU) soon. But It'll take so long time to reproduce on other environments.
I'll try to make this code work robust to the environment.