microsoft/Azure-Kinect-Sensor-SDK #1425: Examples on how to parse MKV files

greatwallet, 2020-11-25:
**Is your feature request related to a problem? Please describe.**

I am a green hand upon video processing and using `ffmpeg` or `mkvtoolnix`. And there is almost no example codes in  documentation, except one or two lines of commands in "https://docs.microsoft.com/en-us/azure/kinect-dk/record-file-format", which only gives commands on getting rgb image via ffmpeg, so it's really painful for me to get stuck into video processing. 

**Describe the solution you'd like**

Would you plz give some example commands of `ffmpeg` and `mkvextract`(`mkvmerge` or something alike) to extract rgb, depth, ir, imu meta and all sorts of data modalities?

**Describe alternatives you've considered**

Maybe in the future, would you work on giving functions on `k4aviewer` so that users can extract any frame or data of any modality on any timestamp via a simple button press?

**Additional context**

Currently I need to extract a huge amount of mkv files and extracts rgb, ir and depth frames, so I want some exact commands so that I can write `cmd` files for batch processing. Currently I could not get ir frames when I tried  
```cmd
ffmpeg -i .\test.mkv -map 0:3 -vsync 0 -vframes 20 test\\ir_test_%04d.png
```
, while the flag for `map` works for depth and rgb when I tried `0:1` and `0:0` respectively.

If anybody gets a clue on that, plz help me out.

vladalx, 2020-11-25:
If you want just one channel extracted, you can try something like this 

ffmpeg -i file_input.mkv -map 0 -map -0:v:1 -map -0:v:2 -c copy file_output.mkv

where :
-map 0              // select all streams
-map -0:v:1         // unselect video stream 1
-map -0:v:2         // unselect video stream 2
( this leaves only video stream 0 to be copied in output )

vladalx, 2020-11-25:
`ffmpeg -i file_input.mkv -map 0 -map -0:v:1 -map -0:v:2 -c copy file_output.mkv`

where :
-map 0              // select all streams
-`map -0:v:1`         // unselect video stream 1
-`map -0:v:2`         // unselect video stream 2
( this leaves only video stream 0 to be copied in output )

vladalx, 2020-11-26:
pls try 
ffmpeg -i input.mkv -map 0 -map -0:v:1 -map -0:v:2 -frames 20 output_%04d.png
( selecting all streams then removing stream 1 and 2, leaving just the rgb )

RoseFlunder, 2020-11-26:
If I remember correctly you could get the IR images with:
ffmpeg -i test.mkv -map 0:2 -vsync 0 ir%04d.png

vladalx, 2020-11-26:
There is an example in the documentation here
https://docs.microsoft.com/en-us/azure/kinect-dk/record-file-format

greatwallet, 2020-11-26:
Got it! Thx everyone!

closed 2020-11-26 by greatwallet

jhljx, 2020-12-30:
So how to extract the IMU data from the mkv file by a ffmpeg command?

`ffmpeg -i test.mkv -map 0:3 -vsync 0 imu%04d.txt` doesn't work. The errors are

    Unknown/unsupported AVCodecID S_K4A/IMU.
    [matroska,webm @ 00000189ba64d6c0] Could not find codec parameters for stream 3 (Subtitle: none): unknown codec
    Consider increasing the value for the 'analyzeduration' (0) and 'probesize' (5000000) options
    [matroska,webm @ 00000189ba64d6c0] Could not find codec parameters for stream 4 (Attachment: none): unknown codec
    Consider increasing the value for the 'analyzeduration' (0) and 'probesize' (5000000) options

It says that the IMU data has an unknown codec.

CcCc-ceil, 2021-07-16:
I have the same question too.

aiwhoo, 2021-12-09:
@RoseFlunder , how would you take the inverse of your ffmpeg call? So instead of breaking the mkv into the three channels, how can we provide the 3 channels (ie ir%04d.png depth%04d.png and color%04d.png in png format and create the mkv?

marcfischer123467, 2022-01-18:
I also tried MKVToolNix:
C:\Program Files\MKVToolNix>mkvextract.exe D:\Desktop\output.mkv tracks 0:color.mkv 1:depth.mkv 2:ir.mkv 3:imu.mkv
with the result:
Error: Extraction of track ID 3 with the CodecID 'S_K4A/IMU' is not supported.

Marselka, 2022-02-01:
I faced the same problem as @marcfischer123467 and other colleagues, and didn't find a solution on how to extract IMU data from mkv. Could one from the DK developers prompt us?

cansik, 2022-09-13:
To extract the calibration data it is possible to use the following ffmpeg command:

```
ffmpeg -dump_attachment:t "" -i OFFICE_SAMPLE_NFOV_2X2BINNED.mkv
```

lijuno, 2023-02-17:
`mkvextract output.mkv tracks --raw 3:imu.bin` dumped IMU data from "output.mkv".

cwidisgroup, 2023-08-08:
How to merge back the extracted frames to a video so that it can be played using Azure Kinect Viewer?

ashutosh3308, 2023-08-08:
How to merge back the extracted frames to a video so that it can be played using Azure Kinect Viewer?

nosignal00, 2024-03-25:
For whom still find the way with ffmpeg
ffmpeg -i inputfile -map 0:3 -c copy -f data IMU.data

You can open the file with hex editor.
The binary format is here.
https://learn.microsoft.com/en-us/azure/kinect-dk/record-file-format#imu-sample-structure