microsoft/Azure-Kinect-Sensor-SDK #1430: AKDK power supply jack (polarity and extension)

SimoSR, 2020-12-02:
I would like o connect he sensor through 2 separate cables (data and power) with a lenght up to 5 meters.

For the data cable USB 3.0 type A i'm going to adopt an active extension cable, but what abut the power supply cable:

1. Can anyone provide the polarity of the power supply jack (barrel and central pin)? 
2. Which the requirements of the USB 2.0 type A extension cable (up to 5 meters) to provide enough power to the device?

qm13, 2020-12-04:
The power jack connector is the same as this: https://www.alibaba.com/product-detail/4506-DC-plug-4-5mm-0_60742810909.html?spm=a2700.7724857.normalList.20.61e761fcUIgZxg

The connector is wired as follows:
[image]

UnaNancyOwen, 2020-12-04:
I can't find a ready-made cable.
We need to buy connectors and cables, and work them ourselves myself.
Does anyone know of a good ready-made cable? Thanks,

mellinger, 2020-12-07:
This is what I use as replacement power cables. https://www.amazon.com/gp/product/B08GLDC587/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

Works with 18v or 20v charger.

closed 2021-02-09 by qm13