microsoft/Azure-Kinect-Sensor-SDK #1431: Get Coordinates of Joints in Unity using C#

ifanroberts99, 2020-12-03:
I am currently trying to track the joints of the user and then output the position while updating from frame to frame. I am using the Microsoft.Azure.Kinect.Sensor and the Microsoft.Azure.Kinect.BodyTracking SDK and I have followed this link to set up my scene:

https://github.com/microsoft/Azure-Kinect-Samples/tree/2a1e75a0ea7a108415b39f9df66cfe09f856029d/body-tracking-samples/sample_unity_bodytracking

So far it seems that opening the device, starting the cameras and the configuration all works which suggests that I have set up the scene correctly but I'm not too sure about the rest. This is my code so far:

[image]

I'm quite new to using C# and the Azure Kinect Sensor and BodyTracking SDKs and was wondering if anyone could help me with this piece of code?

anantley, 2020-12-04:
Hi @ifanroberts99,

The first thing you can try is loading the two sample scenes, PuppetMaster, and Kinect4AzureSampleScene in each of these scenes there is a game object "main" that has a main.cs script and the Kinect4AzureTracker prefab has a TrackerHandler.cs script that handle the body updates.  The puppet scene shows how to apply the joint rotations to a character.

Let me know that doesn't solve your problems

ifanroberts99, 2020-12-04:
Hi @anantley,

This seems to be calling the position of a joint (pelvis). Is that correct?

[image]

If this is true where would I possibly put this code to call the position of another joint and be able to output it? Could I possibly use it here:

[image]

Thank you for all your help so far!

anantley, 2021-01-11:
Hi @ifanroberts99,
I think the easiest thing to do for getting skeleton joints and printing them would be to use the game objects that are under Kinect4AzureTracker->PointBody->(whatever joint you want).   

Any C# script in the scene can be used to access these game objects and print the transform positions, rotations, etc.

closed 2021-02-09 by qm13