microsoft/Azure-Kinect-Sensor-SDK #1435: Order relevant for Camera Synchronisation or only copying timestamp?

Renxiem, 2020-12-07:
Quote from https://docs.microsoft.com/en-us/azure/Kinect-dk/multi-camera-sync
"To get precise image capture alignment between all devices, you have to start the master device last."

I did some recording but accidentally reversed the order when starting the devices, does it remove the whole synchronisation of the devices or just the copying of the master's timestamp to the subordinate devices?

qm13, 2020-12-07:
I am not sure I understand your comment "copying of the master's timestamp to the subordinate devices". The sync cable is analog and transmits a sync pulse to the subordinate cameras. For best results you need to have the subordinate devices stable prior to starting the master camera.

Renxiem, 2020-12-08:
Sorry, I am not very familiar with the synchronisation of devices.
With "copying" i mean when retrieving the device timestamp for e.g. the color image of a subordinate device (when starting it before the master) is similar to the masters device timestamp + set offset while otherwise the device timestamp is completely different for each camera device.

So i can expect the recorded data to still be synchronized but not perfectly?

qm13, 2020-12-09:
Assuming the two Azure Kinects are connected to the same camera then the timestamps for the synchronized frames should not be more than 100us different (they will unlikely be exactly the same).

Technically all subordinates sync to the input pulse in the sync cable. The master is basically free running, which is why it must be started last. All devices reset the timestamp counters at zero and start the timers on the rising edge of the sync pulse. If master is started first, then it will reset and run immediately – it isn’t waiting for anything. Subordinates will then later reset and run, but timers will be completely out of sync, which is why they appear random in your example.

closed 2021-01-08 by qm13