microsoft/Azure-Kinect-Sensor-SDK #1437: PCIe USB connection failure

nictera-ms, 2020-12-09:
Hello,
I am trying to connect to an Azure Kinect DK. All of the USBs on my motherboard are ASMedia, so I installed a Renesas USB3.0 PCIe card (Rosewill RC-508) to try to make this work. I see the device under the correct controller (Renesas USB 3.0 eXtensible Host Controller - 1.0 (Microsoft). This controller is under a PCI-to-PCI Bridge. I see the Kinect Camera, etc where I'm supposed to in the Device Manager and I can use the Kinect Camera with the Windows built-in "Camera" program.

I am getting the following errors from k4aviewer.exe, which shows "(No Available Devices)":

[2020-12-09 11:15:58.842] [error] [t=14312] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (359): find_libusb_device(). Unable to find Container ID: {6fa8ea73-d90a-46c0-b3af-48619bc67dbb} 
[2020-12-09 11:15:58.843] [error] [t=14312] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\color_mcu\color_mcu.c (53): usb_cmd_create(USB_DEVICE_COLOR_IMU_PROCESSOR, NULL_INDEX, container_id, &colormcu->usb_cmd) returned failure in colormcu_create()
[2020-12-09 11:15:58.843] [error] [t=14312] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (152): colormcu_create(container_id, &device->colormcu) returned failure in k4a_device_open()

which are similar to a previous report (https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1086). Alas, if only rebooting could solve this (it doesn't).
Have you had other reports that a USB3.0 PCIe card won't cut it to run Azure Kinect DK with k4aviewer.exe?
Thanks

VitaliyKurokhtin, 2020-12-09:
I had a similar problem earlier today, see if this helps you #1439

nictera-ms, 2020-12-14:
Thank you, @VitaliyKurokhtin. Unfortunately, this did not help.

qm13, 2021-01-08:
This is a know issue on Windows.

qm13, 2021-01-08:
Duplicate of #1409

closed 2021-02-22 by qm13