microsoft/Azure-Kinect-Sensor-SDK #1439: SDK fails to connect to Kinect sensor if same device was earlier plugged into another USB controller

VitaliyKurokhtin, 2020-12-09:
**Describe the bug**

**To Reproduce**
1. On a clean system plug in device into motherboard (ASMedia USB host controller was used)
2. Unplug the device
3. Install supported USB controller card (Speed Dragon EU306A was used)
4. Connect the device to installed controller
5. Run Kinect Viewer

**Expected behavior**
Viewer starts and displays feed correctly

**Logs**
usb_cmd_create(USB_DEVICE_COLOR_IMU_PROCESSOR, NULL_INDEX, container_id, &colormcu->usb_cmd) returned failure
colormcu_create(container_id, &device->colormcu) returned failure in k4a_device_open()
find_libusb_device(). Unable to find Container ID {8358ea0b-ca8f-494c6aa6-2ae38a7528a0}

**Desktop (please complete the following information):**
 - SDK Version:  1.4.1

**Additional context**
This was resolved by going into device manager, turning on hidden devices display and removing any Kinect related hidden device.

czhaneva, 2020-12-21:
Try add "sudo" before your command. e.g. sudo ./k4aviewer
This works for me.

RoseFlunder, 2020-12-21:
@czhaneva usually no sudo is needed. you just have to copy the "99-k4a.rules" files to your "/etc/udev/rules.d/" folder like described here:
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md#linux-device-setup

closed 2021-01-08 by qm13