microsoft/Azure-Kinect-Sensor-SDK #1440: USB-xHCI Driver from Microsoft doesn't work for Azure Kinect

Zailux, 2020-12-11:
**Describe the bug**
I am aware that there are only specific host controller driver working for the Azure Kinect (https://docs.microsoft.com/en-us/azure/Kinect-dk/troubleshooting#usb3-host-controller-compatibility). My desktop ootb uses the ASMedia 3.0 extensible hostcontroller which is already reported not working for the Azure Kinect. Therefore I changed to the Microsoft USB-xHCI-compatible hostcontroller from Microsoft, but the device doesn't start. 

I just need the hint which exact driver i need to get the Azure Kinect running. A download link or the like would be great.
There was no Intel USB 3.0 Host Controller available in the driver list. As far as I googled Intel doesn't even support USB 3.0 Hostcontroller atm.

**To Reproduce**

qm13, 2020-12-14:
The Intel USB Controller driver is included with Windows. You are using an Intel motherboard. Are none of your USB ports backed by an Intel USB controller (integrated with the CPU)?

Zailux, 2020-12-14:
The Intel USB driver are unfortunately only for the USB 2.0 ports. The USB 3.0 drivers are from ASMedia or the other one mentioned above. I have two USB 3.0 Ports which don't have the appropriate 3.0 driver. Can you give me hint where I should find the driver in the device manager menu, or rather the install driver menu ? Better would be a link where I could download the necessary driver, but I guess you guys don't have one directly available. 

Do you need any more specific information, can I provide anything else?

The only driver I found online, are not applicable for my chipset (https://downloadcenter.intel.com/download/22824/Intel-USB-3-0-eXtensible-Host-Controller-Driver-for-Intel-8-9-100-Series-and-Intel-C220-C610-Chipset-Family)

qm13, 2021-01-08:
There is a known issue with ASMedia USB chip sets on Windows. We currently do not have a fix.

qm13, 2021-01-08:
Duplicate of #1409

closed 2021-02-22 by qm13