microsoft/Azure-Kinect-Sensor-SDK #1442: Kinect + Jetson Nano + PoE, any chance of getting it to work?

jackjansen, 2020-12-12:
I'm contemplating investigation of a single-cable solution (for ease of deployment of a multi-camera capture setup).

I was thinking of attaching a Kinect to a Jetson Nano, and adding a power over ethernet shield. I could then have a couple of these around the room (facing inward, towards the subject being captured), with no problems with cable length and such.

The Jetson Nanos would do the capture and preprocessing (and possibly triggering the Kinects, using PTP to sync the Jseton Nanos), do limited compression and send the images to a central machine over Ethernet for further processing.

Is this potentially feasible, or are there any brick walls that I will run into? (I was originally thinking Rpi4, but I've already discovered that the lack of a decent GPU is a brick wall there).

qm13, 2020-12-14:
This should be doable. Verify that you can power the Azure Kinect via TWO USB ports on the Jetson Nano. If that works then no additional power electronics is required. Calculate the max power draw and purchase a PoE splitter for the PoE power category required. You will need an PoE injector or switch/router at the other end.

mellinger, 2021-01-08:
@jackjansen You might want to check out https://github.com/n1ckfg/JetstreamKinect -- I know Nick is working on something similar.

closed 2021-02-09 by qm13