microsoft/Azure-Kinect-Sensor-SDK #1443: Depth alignment to color precision problem

seigeweapon, 2020-12-15:
**Describe the bug**

When making point cloud from azure kinect, I find a precision problem. You can see it from the picture blow. Notice the color of face from one side was mapped to the wall, while on the other side, the wall's color mapped to the face.

**To Reproduce**

My approach to make point cloud: 
1) using depth_image_to_color_camera
2) depth_image_to_point_cloud

**Expected behavior**

transformed depth align perfectly to color image

**Logs**

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS with Version: 
 - SDK Version:  
 - Firmware version:  
Ubuntu 18.04
k4a sdk: 1.3.0
firmware: color:1.6.102 depth:1.6.75 sensor config:6109.7

**Additional context**

seigeweapon, 2020-12-21:
I've done some search and found that similar issue has been reported by quite some users. I think it was discussed pretty thoroughly in #1058 , where @rabbitdaxi describbed the calibration process and possible imperfection in detail. The conclusion seems to be "There's nothing we can do".

However, I think a better calibration may still improve. Hence I'm here to ask the Azure Kinect team to consider open sourcing the calibration process, so that we the user have an oppertunity to fix or at least ease this problem ourselves. Thank you.

seigeweapon, 2020-12-21:
I've done the color to depth transformation too. You can see similar misalignment clearly. So I think there must be some room for improvement in calibration.

[image]

jwvanderbeck-Unity, 2021-01-07:
Running into this same issue and thought it was something I had broken, so sort of glad to see it isn't just me :)  I'm assuming this is a consequence of the depth and color sensors being so incredibly different, even different aspect ratios?

FilipePires98, 2021-03-29:
> I've done the color to depth transformation too. You can see similar misalignment clearly. So I think there must be some room for improvement in calibration.
> 
> [image]

Did you find a way to improve your intrinsics calibration?

meteorshowers, 2021-07-04:
poor device