microsoft/Azure-Kinect-Sensor-SDK #1447: Crash - Tools ImGui backend should use Windows Platform specific renderer (win32 API, DirectX, etc) instead of GLFW

Greendogo, 2020-12-19:
**Describe the bug**

Running the ImGui based tools (k4aviewer.exe for example) on a Windows machine without the appropriate OpenGL version will cause the program to crash with error:

Glfw error [65543]: WGL: OpenGL profile requested but WGL_ARB_create_context_profile is unavailable
Glfw error [0]: glfwCreateWindow failed!

The internet says this is due to using OpenGL 3.1 instead of OpenGL 3.3+.  Many chips make it impossible to upgrade OpenGL when the chip is no longer supported.

**To Reproduce**

Use a "tool" such as "k4aviewer.exed" on an older Windows platform, that otherwise supports the K4A Sensor, but that is running OpenGL 3.1 (this can be CPU dependent).
**Expected behavior**

**Logs**
Produces errors such as:

Glfw error [65543]: WGL: OpenGL profile requested but WGL_ARB_create_context_profile is unavailable
Glfw error [0]: glfwCreateWindow failed!

**Desktop (please complete the following information):**
 - OS with Version: Windows 10
 - SDK Version:  1.4.1
 - Firmware version:  Related to "tool" GUI library, not the hardware.

**Additional context**
Because this is developed by Microsoft, I suggest that for Windows systems you're building ImGui to use GLFW instead of the Win32/directX APIs as these would not cause the crash.

RoseFlunder, 2020-12-19:
I think the depth engine for processing the raw data from the depth sensor of sdk is using open gl too.
So you may have trouble without a proper open gl version nonetheless.
The docs state open gl Version 4.4

Greendogo, 2020-12-31:
The docs state that it requires:
```Graphics driver support for OpenGL 4.4 or DirectX 11.0```

closed 2021-01-08 by qm13