microsoft/Azure-Kinect-Sensor-SDK #1448: Start azure kinect DK at boot

zhaoyuliang5, 2020-12-19:
[2020-12-19 12:57:58.454] [error] [t=9967] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (154): depth_engine_start_helper(). Depth engine create and initialize failed with error code: 204.
[2020-12-19 12:57:58.454] [error] [t=9967] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (160): deresult == K4A_DEPTH_ENGINE_RESULT_SUCCEEDED returned failure in depth_engine_start_helper()
[2020-12-19 12:57:58.454] [error] [t=9967] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (194): depth_engine_start_helper(dewrapper, dewrapper->fps, dewrapper->depth_mode, &depth_engine_max_compute_time_ms, &depth_engine_output_buffer_size) returned failure in depth_engine_thread()
[2020-12-19 12:57:58.454] [error] [t=9882] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (552): dewrapper_start(). Depth Engine thread failed to start
[2020-12-19 12:57:58.454] [error] [t=9882] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth/depth.c (398): dewrapper_start(depth->dewrapper, config, depth->calibration_memory, depth->calibration_memory_size) returned failure in depth_start()
[2020-12-19 12:57:58.455] [error] [t=9882] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (359): cmd_status == CMD_STATUS_PASS returned failure in depthmcu_depth_stop_streaming()
[2020-12-19 12:57:58.455] [error] [t=9882] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (362): depthmcu_depth_stop_streaming(). ERROR: cmd_status=0x00000063
[2020-12-19 12:57:58.455] [error] [t=9882] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (895): depth_start(device->depth, config) returned failure in k4a_device_start_cameras()

RoseFlunder, 2020-12-19:
Pls provide more infos about the way how you start it and on which operating system.

But if its on Ubuntu you will need an active X-Server session for the creation of an open gl context.
Typical system.d autostarts won't work because of that.

zhaoyuliang5, 2020-12-27:
Thanks for your reply.

Ubuntu, Nvidia Jetson XAVIER NX ,   using system Startup Applications Preferences, added a script to start my application. The problem happened  when the DK is working (white light) , NX power off, then restart system(power on), then the DK Depth engine can not be opened again.

qm13, 2021-01-04:
Does the system start ok if you disconnect and reconnect the Azure Kinect device.

closed 2021-01-20 by qm13