microsoft/Azure-Kinect-Sensor-SDK #1457: What are the critical system specifications for a hosting multiple sensors on a single PC?

DrMaxP, 2020-12-28:
I am aware of some people's system specifications for hosting 5-10 Azure Kinect sensors on a single PC - including the DepthKit Ignite demo system https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/971. However, that system had a very high spec and I'm wondering where performance could be lowered without significantly compromising the ability of a single PC to host say 5-7 Azure Kinect sensors capturing RGB data at 1080p and depth at any of the supported FOVs and binning modes. I do not need to do body tracking.

Apart from USB 3 expansion cards that can offer one host controller per sensor, what is most critical? Core count? Memory and speed? CPU max clock speed? Cache size? How important is the GPU if not doing body tracking?

From that thread:

qm13, 2021-01-04:
The most critical resource (since you are not using Body Tracking) is USB bandwidth. You do not mention the frame rates for the data streams. Assuming 30fps you can probably handle 2 streams per controllers and certainly have enough GPU to process your 5-7 streams. Without knowing the compute requirements of your app or DepthKit it is hard to determine if the above HW requirements is sufficient. Since you are using DepthKit I would suggest asking Scatter for guidance.

DrMaxP, 2021-01-09:
Thanks. I'm not using DepthKit. Because I am a hobbyist and can't risk building a high end PC that won't drive multiple Kinects, I've fallen back to the one PC per Kinect option. I already have some very small PCs that were being used to drive Kinect V2s and have determined what characteristics are important to drive an Azure Kinect. These PCs have mostly 4th Gen Intel CPUs and I'm testing Brekel Pointcloud V3 acquiring depth and colour at 30fps. It appears on these older systems you need at least 4 physical cores to do this - i.e. 2 cores with hyperthreading aren't enough on these systems. However, at least this gives me a known route / cost basis for running a multi-sensor system.

DrMaxP, 2021-01-09:
Worth noting I'm not interested in body tracking and therefore the GPU requirements are no where near as important.

closed 2021-01-20 by qm13