microsoft/Azure-Kinect-Sensor-SDK #1459: Help ,Help

UEBoy2019, 2020-12-31:
uint16_t *depth_data = (uint16_t *)(void *)k4a_image_get_buffer(depth_image);
    k4a_float2_t *xy_table_data = (k4a_float2_t *)(void *)k4a_image_get_buffer(xy_table);
    k4a_float3_t *point_cloud_data = (k4a_float3_t *)(void *)k4a_image_get_buffer(point_cloud);

    *point_count = 0;
    for (int i = 0; i < width * height; i++)
    {
        if (depth_data[i] != 0 && !isnan(xy_table_data[i].xy.x) && !isnan(xy_table_data[i].xy.y))
        {
            point_cloud_data[i].xyz.x = xy_table_data[i].xy.x * (float)depth_data[i];

_**//xy_table_data[i].xy.x * (float)depth_data[i]    why it need to * but not point_cloud_data[i].xyz.x = xy_table_data[i].xy.x**_ 

            point_cloud_data[i].xyz.y = xy_table_data[i].xy.y * (float)depth_data[i];
            point_cloud_data[i].xyz.z = (float)depth_data[i];
            (*point_count)++;
        }
        else
        {
            point_cloud_data[i].xyz.x = nanf("");
            point_cloud_data[i].xyz.y = nanf("");
            point_cloud_data[i].xyz.z = nanf("");
        }

qm13, 2021-01-04:
Which sample are you referring too.

mrcrkk, 2021-01-05:
This sample code is from ./examples/fastpointcloud/main.cpp:58

If my understandings are correct, you ask why should x-scale factor be multiplied by the pixel's depth value.
It is needed to get the 3d X-coordinate of a pixel in millimeters.

According to fastpointcloud's [README.md](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/examples/fastpointcloud/README.md):

> At runtime, the 3d X-coordinate of a pixel in millimeters is derived by multiplying the pixel's depth value with the corresponding x-scale factor.

HlibKazakov2000, 2021-01-05:
This sample code is from ./examples/fastpointcloud/main.cpp:58

If my understandings are correct, you ask why should x-scale factor be multiplied by the pixel's depth value.
It is needed to get the 3d X-coordinate of a pixel in millimeters.

According to fastpointcloud's [README.md](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/examples/fastpointcloud/README.md):

>At runtime, the 3d X-coordinate of a pixel in millimeters is derived by multiplying the pixel's depth value with the corresponding x-scale factor.

closed 2021-01-05 by HlibKazakov2000