microsoft/Azure-Kinect-Sensor-SDK #1461: Sample unity body tracking app missing meta files

Utkarsh-Bajpai, 2021-01-04:
When tried to open the PuppetMaster scene in Sample Unity Body Tracking Application, there are missing prefab errors. This is because of missing meta files for the prefabs Female adult and jasper. Meta files are also missing for other files such as all files in the model directory, and a few .cs files.

Could you please add the required meta files.

[image]

anantley, 2021-01-06:
Hi I fixed the problem in .gitignore and will push the fix today

janwww, 2021-01-18:
Hi, I had the same problem and everything seems to work now in your project from your pull request, thanks!

I was wondering if the missing script on the `Kinect4AzureTracker` object in the `PuppetMaster` scene is still relevant (see image).
I couldn't find any scripts with missing `.meta` files and everything seems to work, so I guess it's not important?
[image]

anantley, 2021-01-21:
Hi @janwww,
The missing script is just leftover cruft. You can delete it.

closed 2021-02-09 by qm13