microsoft/Azure-Kinect-Sensor-SDK #1463: Azure Kinect Depth Camera can not be found in device manager

serend1p1ty, 2021-01-05:
**Describe the bug**
After plugging the cables in USB3.0, I only find the `Azure Kinect 4k Camera` in the device manager.
`Azure Kinect Depth Camera` is missing.

In addition, the Kinect is blinking amber, although using USB3.0.

I don't know if there is a relation between the two issues.

**Desktop (please complete the following information):**
 - OS with Version: Windows 1903
 - SDK Version:  1.4.1

qm13, 2021-01-07:
The blinking amber led means there is not enough power to depth camera (hence the depth camera is missing). You do not say but I assume you are plugging the camera into a USB3.0 interface which cannot provide the necessary power. You have two options:
Also connect the included power supply (you may be able to use just the cable plugged into a second USB3 interface on your computer)
Purchase a USBC power+data cable and connect the Azure connect to a USB3.1 (usually uses a USBC female connector) interface. USB3.1 Gen 1 interfaces can provide the necessary power.

serend1p1ty, 2021-01-10:
I have tried both `power caple + data caple` and `USBC power+data caple`, but it still not works.
I contacted the after-sales technical personnel and they said that the problem might be caused by the equipment failure.

serend1p1ty, 2021-01-17:
I bought a new typec usb3.0 data cable, and now the camera is working properly.
Note that the charging cable of most mobile phone and mac is typec usb2.0.

closed 2021-01-17 by serend1p1ty