microsoft/Azure-Kinect-Sensor-SDK #1465: Question about tof type

logic03, 2021-01-06:
https://docs.microsoft.com/zh-cn/azure/kinect-dk/depth-camera
Azure Kinect DK 深度相机实现调幅连续波 (AMCW) 时差测距 (ToF) 原理。 该相机将近红外 (NIR) 频谱中的调制光投射到场景中。 然后，它会记录光线从相机传播到场景，然后从场景返回到相机所花费的间接时间测量值。
请问这里的调幅连续波是指正弦波还是方波呢？因为正弦波一般都是通过相位偏差来间接计算的，但是这里又写是所花费的间接时间。
————————————————————————————————————————————
https://docs.microsoft.com/zh-cn/azure/kinect-dk/depth-camera
Azure Kinect DK Depth Camera implements AMCW Time Difference (TOF) principle.The camera's modulated light in the near-infrared (NIR) spectrum is projected into the scene.It then records an indirect measure of the time it takes for light to travel from the camera to the scene and then back from the scene to the camera.
Is the amplitude modulated continuous wave sinusoidal wave or square wave?Because sine waves are usually calculated indirectly by phase deviation, but here is the indirect time taken.

jaygullapalli, 2021-01-07:
Depth Sensor in Azure Kinect DK uses Indirect Time of Flight Technique. The amplitude modulated wave is a square wave.
You can refer to this paper for additional information https://ieeexplore.ieee.org/document/8310200

logic03, 2021-01-08:
> Depth Sensor in Azure Kinect DK uses Indirect Time of Flight Technique. The amplitude modulated wave is a square wave.
> You can refer to this paper for additional information https://ieeexplore.ieee.org/document/8310200

那请问这个相机的成像传感器上有几个快门呢？是2个用来在不同时刻捕获反射光线吗？
How many shutters are there on the imaging sensor of this camera?Is it two to capture the reflected light at different times?

jaygullapalli, 2021-01-09:
I am not sure I fully understand the question, but the paper referenced above provides bit more insight into the pixel structure (consists of two detectors within the pixel). There multiple phase steps used as part of collecting raw data for computing depth (Phase shift between illumination and pixel clock). You can also refer to this IEEE paper for information about sensor operation https://ieeexplore.ieee.org/document/6964815

closed 2021-01-20 by qm13