microsoft/Azure-Kinect-Sensor-SDK #1469: Calculate space coordinates based on image depth and frustum

xiaoguitoucode, 2021-01-10:
Hi！ I have a question.
I noticed that the FOI of the camera in NVOF mode is 75^*65^. Now I get a depth picture of an object 640*576 (each pixel value represents the depth Z from the camera), now w I want to know the spatial coordinates of a certain point of the object, can I use the angle and depth of the viewing cone Z and the position of the point relative to the depth map, how to calculate its spatial coordinates?
E.g:
*h and *w respectively represent the relative position of the point from the center of the depth map
320 is half the width of the depth map, and 288 is half the height of the depth map
What we require is the spatial coordinates of the point in the camera coordinate system (x, y, z (known))
We can list this formula:
*w/320=x/(z*tan(37.5^))
*h/288=y/(z*tan(32.5^))
The values of x and y can be calculated separately.
Is it right to do this?
thank you !

RoseFlunder, 2021-01-10:
Is there a reason why you don't just use the coordinate transformation functions from the SDK?
https://docs.microsoft.com/en-us/azure/kinect-dk/use-calibration-functions#coordinate-transformation-functions

This function converts the 2D position (x,y) in the depth picture to a 3D coordinate:
https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_ga664602bdb48dab38117a6c1d14b880de.html#ga664602bdb48dab38117a6c1d14b880de

xiaoguitoucode, 2021-01-12:
hi I now use Azure kinect to get some color pictures and depth pictures (mapping depth information 61-90 to 0-255). Is there any way to obtain the spatial coordinates of a certain point offline? I just considered using the above formula, I don’t know if I did it correct.

closed 2021-02-22 by qm13