microsoft/Azure-Kinect-Sensor-SDK #1470: Get all black RGB images in green_screen example

WenTaoLiu502, 2021-01-11:
**Describe the bug**
I tried to run the green_screen example using SDK 1.2.0, the code just displayed all black image in OpenCV window. However, the captured depth images could be displayed in real-time.    

**To Reproduce**
The example was runned on one Azure Kinect device, and the parameters are all default. To reproduce the result I attached part of the code here.

 if (num_devices == 1)
  {
      std::chrono::time_point<std::chrono::system_clock> start_time = std::chrono::system_clock::now();
      while (std::chrono::duration<double>(std::chrono::system_clock::now() - start_time).count() <
             greenscreen_duration)
      {
          vector<k4a::capture> captures;
          // secondary_config isn't actually used here because there's no secondary device but the function needs it
          captures = capturer.get_synchronized_captures(secondary_config, true);
          k4a::image main_color_image = captures[0].get_color_image();
          k4a::image main_depth_image = captures[0].get_depth_image();
          
          // let's green screen out things that are far away.
          // first: let's get the main depth image into the color camera space
          k4a::image main_depth_in_main_color = create_depth_image_like(main_color_image);
          main_depth_to_main_color.depth_image_to_color_camera(main_depth_image, &main_depth_in_main_color);
          cv::Mat cv_main_depth_in_main_color = depth_to_opencv(main_depth_in_main_color);
          cv::Mat cv_main_color_image = color_to_opencv(main_color_image);
          
          // single-camera case
          cv::Mat within_threshold_range = (cv_main_depth_in_main_color != 0) &
                                           (cv_main_depth_in_main_color < depth_threshold);
          // show the close details
          cv_main_color_image.copyTo(output_image, within_threshold_range);
          // hide the rest with the background image
          background_image.copyTo(output_image, ~within_threshold_range);
          cv::imshow("Green Screen.png", output_image);
          cv::waitKey(1);
          
      }
  }

**Expected behavior**
The captured RGB image within the depth threshold could be displayed normally in real-time.

**Logs**
**Screenshots**
RGB image window here:
[image]
Depth image window here:
[image]

**Desktop (please complete the following information):**
 - OS with Version: Ubuntu 16.04
 - SDK Version:  1.2.0
 - Firmware version:  RGB camera firmware: 1.6.98; Depth camera firmware: 1.6.70.

**Additional context**
I also tried to store images captured from two synchronized Azure Kinect devices based on modified green_screen source code. However, the similar issue happened again. The RGB images are all black, and the Depth images are all normal.

qm13, 2021-01-11:
Please update to Ubuntu 18.04. Use k4aviewer to check that you are getting RGB and depth images from the Azure KInect.

WenTaoLiu502, 2021-01-12:
> Please update to Ubuntu 18.04. Use k4aviewer to check that you are getting RGB and depth images from the Azure Kinect.
I have runned k4aviewer before, both RGB & Depth stream are normal. And the  synchronized two Azure Kinect also worked normally on k4aviewer.
[image]
[image]

closed 2021-01-20 by qm13